dcowden/cadquery #213: Documentation: How to save/export as multipart model as DWG and/or SVG

Isstum, 2017-12-30:
Hi there! Great work!

But i miss the part of the documentation where i can read about possible ways to export my generated models.

My plan is to generate a model, using multiple parts and then export the whole model as dwg and svg. 

How do i start?

jmwright, 2017-12-30:
@Isstum Welcome to the community. I'm not sure about multi-part exports, and I'm not near a computer right now, but the tests give a basic SVG export example with one object.

https://github.com/dcowden/cadquery/blob/bdb87b4ed3bdc56251458bb97b7713113b345869/tests/TestCadQuery.py#L78

You should be able to combine all your objects right before you export, but that probably isn't what you're looking for, correct?

Isstum, 2017-12-30:
@jmwright Thank you for the example!

You are right, i must not combine all objects into one.

jmwright, 2017-12-30:
Based on a quick look at the code, it looks like CadQuery is only set up to export one object currently.

https://github.com/dcowden/cadquery/blob/bdb87b4ed3bdc56251458bb97b7713113b345869/cadquery/freecad_impl/exporters.py#L302

I think you'll need to mix in some FreeCAD scripting to achieve what you want. I may get a chance a little later on to prototype some code. What I will probably try is to call `.val().wrapped` on each CadQuery object to get the underlying FreeCAD objects. Then use FreeCAD scripting to collect and export the list of the objects.

jmwright, 2017-12-31:
@Isstum Have a look at the following. I'm not really happy with it since it relies on the FreeCAD GUI, but it's the only way I can get advanced SVG export (multiple objects, colors, etc) to work. I feel like I have a memory of @hyOzd running into the same issue with requiring the FreeCAD GUI when doing STEP export with colors. @easyw may have some thoughts on this too based on his work on automated KiCAD model generation. We try to keep the FreeCAD GUI out of scripts (and definitely out of the core of CadQuery) because the scripts will no longer run headless (on a server with no FreeCAD window displaying).
```python
import cadquery as cq
import FreeCAD
import importSVG
from Helpers import show

# CadQuery code
cylinder = cq.Workplane('XY').circle(15.0).extrude(30.0)
box = cq.Workplane('YZ').rect(40.0, 40.0).extrude(20.0)

# Label the underlying objects so we can identify them later
cylinder.val().label = "cylinder"
box.val().label = "box"

# Display them using the old style Helpers.show code so that the objects show up immediately.
# This allows us to use them below.
show(cylinder, (255, 0, 0, 0.0))
show(box)

# Collect the objects into a list that we can pass to the export function.
# You could also use something like getActiveDocument and getObjects and then
# loop through the objects that are present.
objs = []
objs.append(FreeCAD.getDocument("svg_export_test").getObject("cylinder"))
objs.append(FreeCAD.getDocument("svg_export_test").getObject("box"))

# From drawing mod. Handles colors, multiple objects, etc. The CadQuery export
# function is very limited (one object, I don't think it does colors, etc).
importSVG.export(objs, "/home/jwright/Downloads/test.svg")
```

Isstum, 2017-12-31:
Urgh. This is a real bummer... Because my plan was to run the script on a headless server :(

And i cant get your snipplet running.  First the document want found. I created it:
`NoneType' object has no attribute 'isDerivedFrom'`

But now i got

```
Running the Python command 'CadQueryExecuteScript' failed:
Traceback (most recent call last):
  File "/home/marco/.FreeCAD/Mod/cadquery_module/CQGui/Command.py", line 223, in Activated
    imp.load_source('temp_module', tempFile.name)
  File "/tmp/tmp_Y6d8H", line 41, in <module>
    importSVG.export(objs, "./test.svg")
  File "/usr/lib/freecad/Mod/Draft/importSVG.py", line 1212, in export
    if ob.isDerivedFrom("Part::Feature"):

'NoneType' object has no attribute 'isDerivedFrom'

```
when exporting

jmwright, 2017-12-31:
I developed that script in FreeCAD 0.17. Is that what you're running, or is it 0.16?

There might be a way to modify CadQuery's current SVG export function, but it's not immediately obvious to me how to do that.

jmwright, 2017-12-31:
This is the source code of FreeCAD's export functionality. We could probably adapt it to work for you, eliminating all the GUI-bound checks at the beginning.

https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Draft/importSVG.py#L1203

dcowden, 2017-12-31:
A shape can be multiple objects, by using the compound shape type.  If you try unioning all of your objects, you should get a single object of type compound, which should then work as is with export svg....

Isstum, 2017-12-31:
@jmwright i upgraded to 0.17 and your snipplet works!

@dcowden how does this work? Do you have an example? I would prefer this method if  we can run headless with that!

dcowden, 2017-12-31:
@Isstum I would normally point you to the [examples](https://dcowden.github.io/cadquery/examples.html), but they are currently broken, so i logged  issue #214 for that.

Combining several things into a solid should be as simple as:

```
shape1 = Workplane('XY').circle(2.0).extrude();
shape2 = .... other stuff..
shape3 = shape1.union(shape2)
```
If shape1 and shape2 are both disjoint solids, then shape3 will now be a compound.  You can see that by verifying that shape3.compounds().val() is a single object.

You can't currently change the view angle, but it would be easy to do.  The view vector is currently hard coded, as you can see [here](https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/exporters.py#L256).

We'd happily accept a pull request to add a view vector into the options. Preferably, the view vector would be a 3-tuple and not a freecad vector

jmwright, 2018-01-01:
Thanks @dcowden I didn't know the compound would export that way.

@Isstum the last two lines of this updated script use the compound to export. Will this work for you?
```python
import cadquery as cq
import FreeCAD
import importSVG
from Helpers import show

# CadQuery code
cylinder = cq.Workplane('XY').circle(15.0).extrude(30.0)
box = cq.Workplane('YZ').rect(40.0, 40.0).extrude(20.0)

# Label the underlying FreeCAD objects so we can identify them later
cylinder.val().label = "cylinder"
box.val().label = "box"

# Display them using the old style code so that the objects show up immediately.
# This allows us to use them below.
show(cylinder, (255, 0, 0, 0.0))
show(box)

# Collect the objects into a list that we can pass to the export function.
# You could also use something like getActiveDocument and getObjects and then
# loop through the objects that are present.
objs = []
objs.append(FreeCAD.getDocument("svg_export_test").getObject("cylinder"))
objs.append(FreeCAD.getDocument("svg_export_test").getObject("box"))

# From drawing mod. Handles colors, multiple objects, etc. The CadQuery export
# function is very limited (one object, I don't think it does colors, etc).
importSVG.export(objs, "/home/jwright/Downloads/test.svg")

# Export using compound method as described by @dcowden
compound = box.union(cylinder)
compound.exportSvg("/home/jwright/Downloads/compound_test.svg")
```

Isstum, 2018-01-02:
@jmwright Yes and no.

1. the script does not run outside of freecad. 

`AttributeError: 'module' object has no attribute 'getMainWindow'
`

2. using the compound i dont get colors, just the outline.

@dcowden 
Thank you for fixing the docs! I will work myself through the examples! 

At the moment i am reorganizing my code so i can play around faster and test different ways to export my scetch.

dcowden, 2018-01-02:
Well, I can only take credit for raising the issue. The credit for the fix goes to @adam-unbanzyk

dcowden, 2018-01-02:
Well, I can only take credit for raising the issue. The credit for the fix goes to @adam-urbanzyk

jmwright, 2018-01-02:
@Isstum An option to still run a headless server with my script above would be to use xvfb. I've gotten that working with a [KiCAD file converter I created](https://github.com/jmwright/oshw-code/blob/master/kicad_to_svg_converter/README.md) that has to use UI automation.

Isstum, 2018-01-02:
Ok guys, i think i am on the way. 

Have a look: 
[image]

I got this by using `union(stuff, combine=True, clean=False)` followed by `exportSvg()`

But, as you might see, my cabinet lies on the front. :(

I played around with the viewvector in the file ` /usr/local/lib/python2.7/dist-packages/cadquery/freecad_impl/exporters.py`,  but without any change?! Do i work on the wrong file?

Isstum, 2018-01-02:
@jmwright good idea! I used [pyvirtualdisplay](http://pyvirtualdisplay.readthedocs.io/en/latest/) some time ago for running chromium and selenium on an old headless server. Still working  :D

RustyVermeer, 2018-01-26:
@Isstum , I've made a [branch](https://github.com/dcowden/cadquery/tree/svg_orientation_control) that exposes a view_vector to the user of the ExportSVG function.

Here's a snippet of code that shows an isometric view.

```
import cadquery as cq

# CadQuery code
cylinder = cq.Workplane('XY').circle(15.0).extrude(30.0)
# cylinder.exportSvg("compound_test.svg", view_vector=(1,1,1))
cylinder.exportSvg("cyl.svg", view_vector=(0.612375, 0.612375, -0.5) )
```
which looks like this:

[image]

CAUTION: The vector correctly alters the orientation of the model, but DOES NOT change the XYZ axes in the corner. Those lines are hard-coded into the template, and I ran out of time this evening to come up with a solution to that. Perhaps @jmwright has an idea?

At least with the changes in that branch you might be able to achieve what you need. Or perhaps you've already worked something out for yourself haha.

jmwright, 2018-01-31:
I've dug through our code and I don't see anywhere we're adding the XYZ axes in the corner. I also followed the function calls back through FreeCAD and can't find anywhere to change it. I'm not overly familiar with the FreeCAD codebase though, so I could be missing something.

RustyVermeer, 2018-01-31:
@jmwright it's actually just hardcoded into the SVG template in the exporters module.

[Starting here](https://github.com/dcowden/cadquery/blob/97521c5c46d5e0473b6b3823da3be251feb67387/cadquery/freecad_impl/exporters.py#L381).

It kind of looks like at one point someone drew it manually in inkscape or something and then copied the relevant tags into this template string in the module. That's why I'm not entirely sure what is the best way to transform that axis.

Would it be correct to individually calculate the line start and end points based on the camera's position, or would it make more sense to do it some other way?

dcowden, 2018-01-31:
Hah actually that someone was me.. I originally made this template super fast for showing test results in the examples. I'm glad to see we are past that now.

Im not sure what the end use case @Isstum has in mind, but it might be appropriate to simply remove the axes completely

jmwright, 2018-01-31:
@RustyVermeer Thanks, I totally missed the template.

jmwright, 2018-02-23:
Was cleaning up issues this morning and realized that we never closed the loop on this.

The hard-coded origin/axes marker has been removed in commit d4765b6a6d3dc4a64b84c37ea12fc0bc7a75e99b

@RustyVermeer Would you be willing to create a PR for your `view_vector` change when you get a chance?

RustyVermeer, 2018-03-09:
I think it's fair to close this issue now. The PR is merged and should serve to solve this issue.

closed 2018-03-09 by RustyVermeer

jmwright, 2018-03-09:
Thanks @RustyVermeer