dcowden/cadquery #214: Examples broken

dcowden, 2017-12-31:
The examples are broken with the master branch.

Here's the stack trace:

`Traceback (most recent call last): File "/home/travis/virtualenv/python2.7.14/lib/python2.7/site-packages/cadquery-0.0.0-py2.7.egg/cadquery/cq_directive.py", line 41, in cq_directive exporters.exportShape(result.first_result.shape, "SVG", _s) File "/home/travis/virtualenv/python2.7.14/lib/python2.7/site-packages/cadquery-0.0.0-py2.7.egg/cadquery/freecad_impl/exporters.py", line 67, in exportShape fileLike.write(getSVG(shape.wrapped))TypeError: unicode argument expected, got 'str'`

Does anyone know what broke it?

jmwright, 2017-12-31:
Looks like it could be related to the Python 3 update, but I'm not sure. We should probably think about setting those tests up to be run as a part of the test suite.

dcowden, 2017-12-31:
yeah i agree, they've been broken a few times. I'm not sure ATM how to make sure the docs generate successfully, though.

dcowden, 2017-12-31:
@adam-urbanczyk do you have any ideas about the error above?

adam-urbanczyk, 2018-01-01:
@dcowden  are you running this on py3 or py2?

dcowden, 2018-01-01:
You know, that's a really good question. I don't know!

It's running when Travis runs the sphinxdoc, which uses the default python:

https://github.com/dcowden/cadquery/blob/master/.travis.yml

I don't know what OS Travis uses by default either, so it's possible that it has changed as well.

adam-urbanczyk, 2018-01-01:
Hmm, looking at the traceback I assume it is 2.7.

Note to myself: ```_s``` is io.StringIO so it expects unicode

https://github.com/dcowden/cadquery/blob/53242dd1ded1746631236c825a920c15120de0df/cadquery/cq_directive.py#L37

Trying to fix this by this PR #215. @dcowden could you check if this solves the issue (I am not sure how to reproduce).

dcowden, 2018-01-02:
@adam-urbanczyk udman!

closed 2018-01-02 by dcowden