dcowden/cadquery #221: Documentation update

RustyVermeer, 2018-01-26:


jmwright, 2018-01-26:
> replaced parametricparts.com references with the mybinder Juptyer Notebook link. 

+1  I was going to suggest it anyway.

> perhaps it is time to split the community provided example projects to a new repo? 

Until CQGI, most of the examples lived in the FreeCAD module, which was awkward. I think it was a step forward to move them into the main CQ repo. Breaking them out into their own repo is an interesting idea, but will that be easier than letting people submit a PR to this main repo? Is the issue that the examples get lost in this repo so people don't know about them or don't feel comfortable contributing?

RustyVermeer, 2018-01-26:
I guess it's a bit forward thinking, is all. 

As I was working on the README, I was trying to put the most important info near the top. So, how the user could actually try and use CQ.

The examples fill the README up quite a bit and could be problematic as more come in, as people might not scroll all the way through to read any other important info. 

I don't think it's a pressing thing now, so no need to act on it yet.

dcowden, 2018-01-26:
One benefit of having a separate examples repo is that it can have its own readme-- that's the benefit contemplated here, I think.  There's no argument that we should move examples off the main readme and onto somewhere else.

Another benefit of a separate examples repo for user-contributed examples is that if they are broken or run a long time, cq development is not held up while they get fixed ( because ideally, we don't release cq with broken examples).  I think this is the strongest argument for another repo for user-contributed examples.

I prefer that examples be done in a way that they are rendered from the code, like in the [documentation](http://dcowden.github.io/cadquery/examples.html).  We've caught problems numerous times that would have otherwise gone undetected.  

With the above in mind, one problem with moving the examples to another repo means that repo would need to install cq in order to execute the examples. That would require the examples to use published versions of cq, which has some advantages and some disadvantages. Either way, it would be some work.

I'd suggest for now that we create a contrib folder, and put user-contributed examples there.  This will be simplest, and gives a way for us to NOT run those as a part of the build, and would allow us all to be somewhat more liberal with accepting PRs for that folder.

RustyVermeer, 2018-02-27:
@dcowden @jmwright , should  I make a PR for this, or does it perhaps need some further work / editing?

Willing to do whatever is the better workflow on this.

jmwright, 2018-02-27:
@RustyVermeer My vote would be to go ahead and create a PR so we can start a threaded discussion here about it.

jmwright, 2018-02-27:
@RustyVermeer I just updated the coverage badge at the top of the readme yesterday, so when you create your PR you'll probably see that conflict right away.
Updated line:
```
[![Coverage Status](https://coveralls.io/repos/github/dcowden/cadquery/badge.svg?branch=master)](https://coveralls.io/github/dcowden/cadquery?branch=master)
```
I think that's the only change since you forked the documentation, but I could be missing something.

RustyVermeer, 2018-03-09:
closing this as the PR has been accepted.

closed 2018-03-09 by RustyVermeer