dcowden/cadquery #233: Python3 STL export stuck and eating resources

Grawp, 2018-03-02:
Hi
I can't export a simple box with a hole. [Here](https://gist.github.com/Grawp/ba2acae074bb42fbac9d5bf3afe708f8) is the script I'm running with python3.

It gets stuck? after stepping into https://github.com/dcowden/cadquery/blob/583d837c1c5bfcef7e4cf3c3f1401bcd578ce050/cadquery/freecad_impl/exporters.py#L87
I can't even debug it with PyCharm or cmdline `pdb`. It just start taking whole CPU core and memory usage start rising (Over 2GB after few minutes and continuing...)

I'm using FreeCAD 0.17 (9ea89f5b5) with python3. (Built in Archlinux from my AUR [package](https://aur.archlinux.org/packages/freecad-python3-git/).)

STEP export works fine. I can open exported file with the FreeCAD just fine.

Any ideas?

jmwright, 2018-03-02:
@Grawp Does the object export to STL fine if you use the File menu in the FreeCAD GUI?

Grawp, 2018-03-02:
I haven't tried `cadquery-freecad-module` since I started using 0.17 with python3. Going to try now.

jmwright, 2018-03-02:
I don't think I've updated the CadQuery lib in the module yet. I can do that a little later, but I'm also curious if your installation of FreeCAD 0.17 can do the STL export without CadQuery involved.

Grawp, 2018-03-02:
I've just successfully exported a STL from my FreeCAD and verified its validity in other applications.

Just FIY: cadquery-freecad-module failed to load with. (probably not surprising?)

> During initialization the error name 'unicode' is not defined occurred in /home/michal/.FreeCAD/Mod/cadquery-freecad-module/Init.py                           
Please look into the log file for further information                                                                                                         
During initialization the error No module named 'ExportCQ' occurred in /home/michal/.FreeCAD/Mod/cadquery-freecad-module/InitGui.py                              
Please look into the log file for further information

Grawp, 2018-03-02:
I really wanted to use python3 but I can try the same script with python2 and stable FreeCAD 0.16 and if it's working, retry it with python2 and the same revision of FreeCAD 0.17 I'm currently using but built with python2 stuff.

jmwright, 2018-03-02:
How much of a pain would it be to get the Python 3 FreeCAD installation running if I have a running Arch VM?

Grawp, 2018-03-02:
Python2 with FreeCAD 0.16.6712 and CadQuery 583d837c1c5bfcef7e4cf3c3f1401bcd578ce050 have the same problem. The script above gets the python interpret somehow stuck on that method.

The Python 3 FreeCAD should be easily installable by `trizen -Sy freecad-python3-git` or `yaourt -Sy freecad-python3-git`. If it is not so, please report a bug :D

But since it's not working even with python2 and freecad from official Arch repos I guess it won't help much. It may still be Archlinux related problem though. I should probably try it in Ubuntu.

Grawp, 2018-03-03:
The script works with python2 and FreeCAD 0.16 in Ubuntu :/

Grawp, 2018-03-03:
This seems to be related https://forum.freecadweb.org/viewtopic.php?f=22&t=23143

jmwright, 2018-03-03:
I had an issue yesterday with the Python 3 Travis environment where some exports would hang. I have weird issues with Travis fairly often, so I didn't think much of it. It must be related to the issue you're having.

Grawp, 2018-03-03:
The [workaround](https://forum.freecadweb.org/viewtopic.php?f=22&t=23143) seems to be working
https://gist.github.com/Grawp/6f2c1d228cc6e94c842509a224297b39

adam-urbanczyk, 2018-03-03:
I can also reproduce this issue (on `freecad                   0.17                    py36_11    freecad`):

```python
import cadquery as cq

box = cq.Workplane("XY").box(15, 15, 70).faces(">Z").workplane().hole(10, 60)
result = box
result.val().wrapped.exportStl('out.stl')
```

It hangs inside FreeCAD code to my understanding, so I am not sure how to approach this.

Grawp, 2018-03-03:
I'll probably build a debug version of FreeCAD and set up a debugging with QtCreator. I tried cmdline gdb but since I don't know anything about FreeCAD's code just simple backtrace without visually going through the code in IDE tells me nothing.

adam-urbanczyk, 2018-03-03:
@Grawp  this works for me (I specified the deflection explicitly):

```python
import cadquery as cq

box = cq.Workplane("XY").box(15, 15, 70).faces(">Z").workplane().hole(10, 60)
result = box
result.val().wrapped.exportStl('out.step',0.01)
```

So the workaround is to specify deflection. If not specified it is assumed to be 0 in FreeCAD code: https://github.com/FreeCAD/FreeCAD/blob/76264ec1c76eddc990ad20e96ec81b21b234b4ef/src/Mod/Part/App/TopoShapePyImp.cpp#L557 .

adam-urbanczyk, 2018-03-03:
@Grawp  could you verify whether #234 solves your issue?

Grawp, 2018-03-03:
@adam-urbanczyk Yes it does work. :)

jmwright, 2018-03-03:
@adam-urbanczyk Thanks for fixing that. I've merged your PR into master.

Grawp, 2018-03-03:
I guess I can close this now. Thanks!

closed 2018-03-03 by Grawp