dcowden/cadquery #238: 2d planar wire extrusion on coincident start & finish

fragmuffin, 2018-03-10:
### Problem

2d wires with coincident start & finish points cannot be generated without using `.close()` to create the last linear wire.

Using `.close()` imposes a mandatory linear line somewhere on the wire.

### Positive Test

The following code functions as I'd expect; extruding a triangle into a 3d solid

```python
import cadquery
from Helpers import show

obj = cadquery.Workplane('XY') \
    .moveTo(0, 0).lineTo(1, 0).lineTo(0, 1) \
    .close(0, 0) \
    .extrude(1)

show(obj)
```

[image]

### Reproducing Erroneous Behaviour

The following are small changes to the above definition of `obj`

**1) Manually closing yields no solid**

```python
obj = cadquery.Workplane('XY') \
    .moveTo(0, 0).lineTo(1, 0).lineTo(0, 1) \
    .lineTo(0, 0) \
    .extrude(1)
```

no solid is generated

**2) Closing on coincidental point**

```python
obj = cadquery.Workplane('XY') \
    .moveTo(0, 0).lineTo(1, 0).lineTo(0, 1) \
    .lineTo(0, 0).close() \
    .extrude(1)
```

raises exception: `Line through identic points`

### Impossible Solid?

There are usually ways to get around this limitation, but I found the following solid impossible to create

[image]

I ended up [creating it in 2 halves, then `union` both sides](https://fragmuffin.github.io/cqparts/doc/cqparts_fasteners/biscuit/component-parts.html#biscuit).

dcowden, 2018-03-10:
Can't find it now, but there has been a lot of discussion on this. It came up when working on dxf import.... It's desirable to get the code the creates the wires from edges separated . 

Last I remember, I think I decided that instead of pushing wires onto the pending wires list, we should just collect all of the edges together and form wires in extrude.

fragmuffin, 2018-03-10:
I imagine any re-design of forming 2d wires will impact #172. Perhaps this could be the first step toward parametric 2d sketches.

RustyVermeer, 2018-03-10:
I often end up doing the same thing. It's a workaround in my opinion.

I think that it would be useful to add logic to close(). If the set of edges / wires is open, then add a line to close it, otherwise, just return a single wire and work with that directly.

It might be as simple as checking if start point = end point and doing nothing if that's true...

Or maybe it's way more difficult. That's just a first thought

fragmuffin, 2018-04-29:
As agreed in [#257#issuecomment-385205989](https://github.com/dcowden/cadquery/pull/257#issuecomment-385205989), test 1 is not a failed test-case.
So after #257 is merged, I think this issue can be closed

gntech, 2018-04-29:
#257 have been merged now

closed 2018-04-29 by fragmuffin