dcowden/cadquery #240: Type hints (PEP-484)

Grawp, 2018-03-12:
Would somebody be interested in adding type annotations so we could have functioning type hinting in IDEs like PyCharm and possibility of using mypy for static analysis?

jmwright, 2018-03-12:
I'm not opposed to this being done, but I personally don't have any room on my CadQuery to-do list for it.

fragmuffin, 2018-03-13:
Just a thought: could this be automated form the existing sphinx-compatible docstrings?
eg: https://pypi.python.org/pypi/sphinx-autodoc-napoleon-typehints

Grawp, 2018-03-14:
@jmwright I've been just adding methods' return values for now but it already has made PyCharm's auto-completion, argument type-hinting etc.. much more better.

I don't know if is possible to have complete internal type annotation so we can run mypy on CadQuery itself but I'll publish .pyi stubs for the API on my github if/when I get it to more complete state.

Grawp, 2018-03-18:
I've started building typed stubs of the API: https://github.com/Grawp/cadquery_stubs

jmwright, 2018-03-21:
@Grawp Forgive my ignorance. What do we do with pyi files once you've got them ready? Do we just put them in the same directory as their counterpart .py and .pyc files?

fragmuffin, 2018-03-21:
@Grawp and another question:
Is there a way to unit test the associated types with each function.

It would be good to know that if a function signature changes but a .pyi file does not, that we'd have a failed test to trigger a change.

Grawp, 2018-03-25:
@jmwright Sorry for a late answer. When (if ever) Python 2 support in CadQuery gets deprecated, you can copy the signatures out from the stubs and put them directly to cq. I've sort-of-just-now realized that the type hints are not a valid Python 2 syntax. So there is probably no action to be done from CadQuery project's side.

If your question was more aimed at "How can I use them?", then

- If you use JetBrains PyCharm, place them anywhere where you would place any other module (e.g. directly into your project folder) so the IDE sees two CadQuery modules .. one with real code and one with the stubs. Enjoy better auto-completion, parameters hinting, etc.
- Use mypy to verify your fully type annotated cq project. See https://mypy.readthedocs.io

@fragmuffin PEP 484 are just hints for IDEs and static analyzers (and I've also seen a run-time assertion framework in development) without any meaning for the Python interpret itself. To ensure they match reality we would need to fully type annotate CadQuery internals and run mypy analysis on the CadQuery itself.

Since the type annotation are not valid Python 2 code we can probably close this issue for now?

jmwright, 2018-03-26:
@Grasp I'm wondering if we should add a link to your cadquery_stubs repo to the docs somewhere before we close this issue. I'll have to put some thought into where it would fit, and am open to suggestions.

Grawp, 2018-04-06:
@jmwright I've got no idea what would be the best place. Anyway when I have some time I'll try to complete the stubs and play with mypy a little to see whether it would be realistic to do complete static analysis (to have some guarantes that the API signatures are right).

jmwright, 2018-05-11:
@Grawp I haven't figured out where to put this in the CadQuery documentation yet, but I mentioned it in the reworked FreeCAD module documentation [here](https://github.com/jmwright/cadquery-freecad-module/blob/master/docs/usage.md#using-an-external-code-editor).

closed 2018-12-06 by dcowden