dcowden/cadquery #243: Version 1.1.0 Released

jmwright, 2018-03-16:
I've posed a release announcement on the Google Group.

https://groups.google.com/forum/#!topic/cadquery/EQGqR86WqxI

Thank you to all who have contributed. Adding Python 3 support was a big milestone for us, but there are a lot of other things packed into this release as well.

fragmuffin, 2018-03-17:
when installing with `pip3` (ie: python3's `pip`) I get:

```
The directory '/home/tester/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/tester/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting cadquery (from -r /code/src/cqparts/requirements.txt (line 1))
  Downloading cadquery-1.1.0.tar.gz (94kB)
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-3yq6ih8l/cadquery/setup.py", line 32, in <module>
        long_description=open('README.md').read(),
      File "/usr/lib/python3.5/encodings/ascii.py", line 26, in decode
        return codecs.ascii_decode(input, self.errors)[0]
    UnicodeDecodeError: 'ascii' codec can't decode byte 0xc5 in position 10630: ordinal not in range(128)
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-3yq6ih8l/cadquery/
```

is the `setup.py` python 3 compliant?

fragmuffin, 2018-03-17:
Oh, it seems it's an issue with the `u'ń'` in your name @adam-urbanczyk in the readme: [Moving to Python3](https://github.com/dcowden/cadquery#moving-to-python3-and-away-from-freecad-as-a-dependency)

fragmuffin, 2018-03-17:
This error is not replicated when running in a `virtualenv`:

**Does NOT reproduce problem**

```
$ mkvirtualenv --python=$(which python3) py3
(py3) $ pip3 install cadquery
```

I believe `setup.py` is run by `$(which python)` which is usually python2 on most systems.

I've produced this error in a docker environment with **only** a `python3` installation.

**Reproduce**

To reproduce you can run my `cadquery` release testing by first building the docker environments

```
git clone git@github.com:fragmuffin/cqparts.git
cd cqparts

# build base environment
pushd env/ubuntu-py3
./build.sh
popd

# build deployment test environment
pushd deployment/env/ubuntu-py3
./build.sh
popd

# create a container
cd deployment
./deploy.sh env new ubuntu-py3
docker exec -it -u root deploytest bash

# from inside the docker container (as root)
pip3 install cadquery
```

adam-urbanczyk, 2018-03-17:
@fragmuffin  I cannot reproduce your error in an anaconda env using python3.6 and pip.

What is `pip3` precisely? I assume it is a distribution specific thing. Could that be the root cause?

```
pip install cadquery==1.1 
Collecting cadquery==1.1
  Downloading cadquery-1.1.0.tar.gz (94kB)
    100% |████████████████████████████████| 102kB 917kB/s 
Requirement already satisfied: pyparsing in ./anaconda/lib/python3.6/site-packages (from cadquery==1.1)
Building wheels for collected packages: cadquery
  Running setup.py bdist_wheel for cadquery ... done
  Stored in directory: /home/X/.cache/pip/wheels/65/ab/5f/ebb5d86563cf7f42e3bc26dfdb96467fca15d00b0e54213c3e
Successfully built cadquery
Installing collected packages: cadquery
Successfully installed cadquery-1.1.0
```

fragmuffin, 2018-03-18:
@adam-urbanczyk 
`pip3` is just the python 3 version of pip, it ensures it's run in python 3 by explicitly setting the runtime binary at the top of the file... for example:

```
$ head -n 1 $(which pip3)
#!/usr/bin/python3
```

**Encoding & Locale**

I think I've found the root cause:

```
# the trigger is the code
open('README.md').read()

# which, as I understand is roughly equivalent to
open('README.md', encoding=locale.getpreferredencoding(False)).read()
```

The encoding of the file deaults to that of the environment's locale.
The erroneous system has a default encoding of: `ANSI_X3.4-1968`, but for most systems the default encoding is 'UTF-8'.
The format of the `README.md` is `UTF-8`, this is irrespective of which system it's decoded on, so the `open()` command should explicitly define the file's encoding.... fix submitted in #244

adam-urbanczyk, 2018-03-18:
@fragmuffin  good catch! I'd use `io.open(...,encoding='utf-8')`  , but your proposal is equally good.

fragmuffin, 2018-03-18:
@adam-urbanczyk `io.open` looks much cleaner, I'll use that in future.
If we remember, we could swap out `codecs` when we next change `setup.py`...

jmwright, 2018-04-10:
I think we're all good on this. Thanks to @fragmuffin and @adam-urbanczyk for sorting things out.

closed 2018-04-10 by jmwright