dcowden/cadquery #246: CI on Windows

adam-urbanczyk, 2018-03-18:
It would be nice to test on Windows too. @dcowden @jmwright  could you add this project to https://www.appveyor.com/ ?

adam-urbanczyk, 2018-03-20:
@dcowden @jmwright  any thoughts on appveyor? I will try it on my fork. I see they have conda preinstalled BTW.

dcowden, 2018-03-20:
I know little about it so I'm ok with it if you are...

jmwright, 2018-03-21:
@adam-urbanczyk I'm in the same situation as @dcowden but I'm supportive of trying it. I linked my GitHub account to AppVeyor and set a CadQuery project up (I think). @adam-urbanczyk you were added as a collaborator with administrator privileges, and @dcowden you were added as a team member with admin privileges.

dcowden, 2018-03-21:
Thanks @jmwright, got the email notification...

adam-urbanczyk, 2018-03-21:
@jmwright I go the notification too. Not sure what to do with it / how to access the project. I will try first to setup things on my fork using conda / python3.6

adam-urbanczyk, 2018-03-27:
Solved by #248

closed 2018-03-27 by adam-urbanczyk