dcowden/cadquery #252: Create arc using start, end and sagitta

gntech, 2018-04-18:
I want to propose an additional convinient way to create arcs beside threepointarc.

Using start point, end point and sagitta. See [Sagitta on wikipedia](https://en.wikipedia.org/wiki/Sagitta_(geometry)). Using either a positive or negative sagitta value makes the arc go either way of the reference line between start and end point.

Do this functionallity already exist for cadquery? Otherwise I can volunteer to add it.

I have already implemented this in a Freecad python script which I think I could quite easily adapt for cadquery.

dcowden, 2018-04-18:
@gntech We do not have this feature and would love the contribution!

Please include a test case and an example in addition to the cause changes.

jmwright, 2018-04-18:
Thanks for the contribution @gntech !

Looks like Appveyor build failed because of a bad gateway error for one of the packages.
```
CondaHTTPError: HTTP 502 BAD GATEWAY for url <https://conda.anaconda.org/conda-forge/noarch/pytz-2018.4-py_0.tar.bz2>
Elapsed: 00:10.942039
CF-RAY: 40da41fbaba09eed-ORD
```
I can download the package directly from the URL, so I restarted the build. It seems to have worked the second time.

@dcowden Will you have time to review and merge, or should I?

dcowden, 2018-04-18:
@jmwright I've got this one, will do it tonight