dcowden/cadquery #261: Release 1.2

gntech, 2018-05-03:
Is there any current plan for when cadquery v1.2 will be released? Are there any features or bug fixes pending?

The reason I am asking is that I would like to have the new radiusArc and sagittaArc functions available by default in the freecad workbench and that seems to use only the released version of cadquery.

Personally I have symlinked in the latest cadquery git repository code to my install of the freecad module but I would like to share my scripts (that uses the new arc functions) to people without the hassle of them needing to do that procedure.

jmwright, 2018-05-03:
@gntech Please update the workbench (easiest to do it through the addons manager). I've updated the workbench master to include the latest CadQuery commits. Please let me know if your changes test ok with it.

gntech, 2018-05-05:
I tested this today but it did not work. Then I saw that you had reverted that commit on the freecad-module. Have you discovered anything wrong? Is there something I can assist with?

jmwright, 2018-05-05:
@gntech What didn't work? Did it say it couldn't find cqparts to import? I reverted the cloned version of cqparts in the repository and now have a script that just copies in the relevant parts manually.

One thing that could potentially keep cqparts from working in the FreeCAD module is if its embedded copies of cqparts are overridden by the ones installed by pip. We've had issues before where somebody had an older version of CadQuery installed and so they weren't getting the benefit of the updated version embedded with the FC module.

gntech, 2018-05-05:
No, I was referring to cadquery. it was the sagittaArc functions in cadquery that was not available.

gntech, 2018-05-05:
@jmwright I am sorry if I am unclear, but what I meant was that you first updated cadquery in the freecad-module after my request in this issue. See commit https://github.com/jmwright/cadquery-freecad-module/commit/19994ffd488224d0eb9540e083a7cbeab99d4db7

But then you have reverted that commit here. https://github.com/jmwright/cadquery-freecad-module/commit/92b07d059d58b3278f110403d838d026728ba9ea

jmwright, 2018-05-05:
> This should be unrelated to cqparts as far as I understand it.

@gntech It shouldn't be related to cqparts, but it is because I made a mistake. When I reverted to an earlier commit to try to remove the cloned version of cqparts, I accidentally reverted the updates to the embedded cadquery library without realizing it. I just fixed that and tested installing it using the addons manager in FreeCAD 0.17. Example 005 that you made changes to seemed to work fine.

If I have time I may work on putting a 1.2 release together tonight (US time), but your scripts should work immediately once you update `cadquery_module` through the addons manager. Please let me know if you continue to have trouble. Sorry for the confusion. I was hopping between discussions on three projects this morning and misinterpreted what this one was related to.

gntech, 2018-05-05:
@jmwright you are awesome, thank you very much! :) Now I can create my models again! :+1: 

> I was hopping between discussions on three projects this morning and misinterpreted what this one was related to.

Yes I have seen that there have been very high activity in both cadquery, cqparts and freecad-cadquery-module lately! Very inspiring discussions, I think all these projects have a bright future ahead. Very much right in time and right technology level in my opinion.

jmwright, 2018-05-06:
I've updated the [change log](https://github.com/dcowden/cadquery/blob/master/changes.md) to reflect the contributions made since the last release. This is being done in preparation for the next release (1.2.0). If anyone wants to take a look, please let me know if anything is missing or incorrect. I'll probably do the release in the next couple of days.

jmwright, 2018-05-07:
@dcowden The Travis-Sphinx build seems to be broken. I'm not sure if there was a recent update to their repos or what. Here's the error I get.
```
The command "coverage run runtests.py" exited with 0.
0.23s$ if [[ "$TRAVIS_PYTHON_VERSION" == "2.7" ]]; then travis-sphinx build --nowarn --source=doc; fi
Traceback (most recent call last):
  File "/home/travis/virtualenv/python2.7.14/bin/travis-sphinx", line 7, in <module>
    from travis_sphinx.main import main
  File "/home/travis/virtualenv/python2.7.14/lib/python2.7/site-packages/travis_sphinx/__init__.py", line 3, in <module>
    from . import build
  File "/home/travis/virtualenv/python2.7.14/lib/python2.7/site-packages/travis_sphinx/build.py", line 7, in <module>
    from sphinx.cmd.build import build_main
ImportError: No module named cmd.build
The command "if [[ "$TRAVIS_PYTHON_VERSION" == "2.7" ]]; then travis-sphinx build --nowarn --source=doc; fi" exited with 1.
```

dcowden, 2018-05-07:
that's pretty weird. I haven't see that before. i think we should start by upgrading to the latest travis-sphix module ( probably the version we're using is frozen inside of requirements-dev.txt )

jmwright, 2018-05-08:
travis-sphinx isn't version locked, but Sphinx and sphinx-rtd-theme are. travis-sphinx probably just got upgraded and it broke backwards compatibility somewhere. I'll tinker with travis-sphinx's version tonight to see if I can fix it.

```
sphinx-rtd-theme==0.1.9
travis-sphinx
Sphinx==1.3.2
coverage
coveralls
pyparsing
mock
codecov
```

jmwright, 2018-05-08:
Yeah, there were two new releases yesterday...

https://pypi.org/project/travis-sphinx/#history

jmwright, 2018-05-08:
Ok, locking it at the version we've been running on since February fixed it. Thanks for the heads-up.

jmwright, 2018-05-09:
There's an official release out there for v1.2.0 now. Release announcement [here](https://github.com/dcowden/cadquery/issues/262).

@dcowden @adam-urbanczyk Travis CI is still being very finicky. About every other commit it fails because of a download timeout error or breaks for some reason unrelated to the code that was just merged/pushed. I don't know what to do about it, but I just wanted to make sure that you were aware that there are issues.

closed 2018-05-09 by jmwright