dcowden/cadquery #263: Shell Broken for Loft Objects

jmwright, 2018-05-09:
This issue began over on the Google Group [here](https://groups.google.com/forum/#!topic/cadquery/E0_0av34rbY). I'm posting it here so we can keep track of it.

Shelling works fine with the [cube example in the documentation](https://dcowden.github.io/cadquery/selectors.html#combining-selectors).
```python
result = cq.Workplane("XY").box(2, 2, 2) \
    .faces(">Z") \
    .shell(-0.2)
```
But no combination of face selections or shell thicknesses seems to be able to make this work.
```python
import cadquery as cq
from Helpers import show

result = cq.Workplane("front").circle(1.5) \
.workplane(offset=3.0).rect(0.75, 0.5).loft(combine=True) \
.faces("-Z").shell(-0.1)

show(result)
```
I used the old `show` function in that example because it provides a stack trace, whereas CQGI's `show_object` function does not.

Here's the main error I get.
```
BRep_API: command not doneRunning the Python command 'CadQueryExecuteScript' failed:
Traceback (most recent call last):
  File "/home/jwright/.FreeCAD/Mod/cadquery_module/CQGui/Command.py", line 226, in Activated
    imp.load_source('temp_module', tempFile.name)
  File "/tmp/tmphTKlY4", line 6, in <module>
    .faces("-Z").shell(-0.1)
  File "/home/jwright/.FreeCAD/Mod/cadquery_module/Libs/cadquery/cadquery/cq.py", line 820, in shell
    s = solidRef.shell(self.objects, thickness)
  File "/home/jwright/.FreeCAD/Mod/cadquery_module/Libs/cadquery/cadquery/freecad_impl/shapes.py", line 1009, in shell
    return Shape.cast(self.wrapped.makeThickness(nativeFaces, thickness, tolerance))

BRep_API: command not done
```

adam-urbanczyk, 2018-05-10:
Looks like a problem with the cad kernel. Did you try with a different loft based model?

jmwright, 2018-05-10:
@adam-urbanczyk  No, I only tried this one.

closed 2018-12-06 by dcowden