dcowden/cadquery #267: suppress_stdout_stderr seams to suppress to much

gntech, 2018-05-16:
When exporting a STEP-file cadquery swallows some of the things written to stdout.

Consider the following script. Run from terminal with python.

``` python
import cadquery as cq

print("before creating model")

m = cq.Workplane("XY").box(10, 10, 10)

print("after creating model and before exporting model")

with open("testfile.step", "w") as f:
    cq.exporters.exportShape(m, "STEP", f)

print("after exporting model")

```

I would expect to get to get 3 print statements but I only get the last one `after exporting model`

I suspect that this has something to with the function suppress_stdout_stderr that is designed to suppress some messages from freecad regarding the step-export.

If I run the script like so instead I get all 3 print statments but also some messages from freecad regarding step creation. Using this form of the export I bypass the suppress_stdout_stderr()

``` python
import cadquery as cq

print("before creating model")

m = cq.Workplane("XY").box(10, 10, 10)

print("after creating model and before exporting model")

m.val().exportStep("out.step")

print("after exporting model")

```

My suggestion for how to fix this issue is to contact freecad upstream to see if we can get a clean step output from freecad to get rid of suppress_stdout_stderr. Because, seriously, the suppress_stdout_stderr() is not the ideal solution. :)

fragmuffin, 2018-05-17:
It seems strange that you're missing messages _before_ you run the exporter.
Is this because `print` buffers output for a short while before flushing them?
Could the fix involve flushing the buffer `print` uses?

> My suggestion for how to fix this issue is to contact freecad upstream [...] because, seriously, the suppress_stdout_stderr() is not the ideal solution. :)

totally agree :+1:. Those wacky FreeCAD people!

gntech, 2018-05-17:
> It seems strange that you're missing messages before you run the exporter.
Is this because print buffers output for a short while before flushing them?

Yes, this is what I think as well. It is bad since I want to use the print statements to show progress of the script. And the buffering makes all the print come at the very end of the script run, so much for progress. :)

gntech, 2018-05-17:
I found a possible solution to the problem that freecad buffers everything that is written to stdout until the end of execution.

`sys.stdout.flush()`

See https://forum.freecadweb.org/viewtopic.php?f=22&t=28354&hilit=Step+stdout

fragmuffin, 2018-05-17:
@gntech `sys.stdout.flush()` worked for me!
Adding it to the start of [`suppress_stdout_stderr.__enter__`](https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/__init__.py#L44) should do the trick.

@adam-urbanczyk @dcowden 

> [adam-urbanczyk 's] branch had just about everything working without FreeCAD

That's exciting! @adam-urbanczyk is it time to start migrating `cqparts`' freecad dependencies out?

dcowden, 2018-05-17:
@adam-urbanczyk @fragmuffin  it is definitely close enough i think its time to try to make it happen. I think he reported having all of the tests passing at the time even. Our test coverage is pretty good, so I think that means we're pretty close.

I've created #268 to get the ball rolling on that.