dcowden/cadquery #268: Move to OCC instead of FreeCAD

dcowden, 2018-05-17:
@jmwright @adam-urbanczyk @fragmuffin 

@adam-urbanczyk has a branch that moves most CQ functionality to use OCC directly, rather than the FreeCAD layer.

I'd like to begin the process of moving that into CQ main version.

Opinions on how to manage the reverse compatibility are welcome. For now, we'll probably use the OCC that comes with FreeCAD already.

My thinking is that we make an initial move, and then slowly move features from FC to OCC, rather than a huge push that changes it all, but what do you guys think? Maybe we should just move it all since @adam-urbanczyk has most if not all tests passing?

osterwood, 2018-05-17:
I've had good success using Adam's OCC branch.  As the code base is merged in, you may want to be aware of this patch I made to support cqparts' GLTF exporter:
    https://github.com/CapableRobot/cadquery/commit/eae28b8dddeeed096a895512880fc50aa6ea3857

dcowden, 2018-05-17:
@osterwood thanks for the information!
@adam-urbanczyk  @jmwright  do you have any reservations about making adams branch a real boy? If the tests pass, i'd be ok doing a new release for it.

@fragmuffin @jmwright  from a cqparts viewpoint, and cqfm viewpoint, i would think a two step process would be best-- first, a new OCC release that seems to work, and then after that changes to cqparts and cqfm to use the new one?

adam-urbanczyk, 2018-05-17:
@dcowden  I am in favor. 
@osterwood  looks interesting, I'll try to merge it if you don't mind. Why is numpy actually needed?

AFAIK there is no PythonOCC bundled with FreeCAD (at least not on my current machine). FreeCAD has a handwritten wrapper for the OCC kernel. I think what we need to begin is an API for selecting the impl module.

osterwood, 2018-05-17:
numpy is just for the matrix convenience methods (shape, transpose, flatten).  The logic could easily be embedded within the file.  The `_print` methods can be ignored -- they helped with debugging but are no longer called.

`transposed_list()` is a surrogate for cqpart's call to `transposed().A` in the GLTF exporter.  I forked that as well and added a fallback call, see https://github.com/CapableRobot/cqparts/commit/4c9f1c804e296176859cd8a549ac718fd566dfe0

Definitely not the cleanest way to "decouple" cqparts from the underlying implementation within cadquery.  I made the same sort of change here https://github.com/CapableRobot/cqparts/commit/58d237fee4879406247aa04cca6bd82a17847741 for the shape exporters -- having cqparts call the cadquery export methods, if the existing & direct call to freecad fails.

dcowden, 2018-05-17:
@adam-urbanczyk oh, i didnt know that your OCC branch was based on a freecad-specific OCC wrapper.

How much flexibility do we gain moving from FreeCAD objects to FreeCAD-authored OCC wrapper objects, vs moving to pythonOCC?

I have a love-hate releationship with pythonOCC. I love the development speed, the breadth of the project, and the people ( Jelle and Thomas), but I've historically found developing with it difficult. But its typically not been pythonOCC's fault-- its been OCC itself: its a bear to use at the base level.

I know for sure I need the direct OCC objects to do the historical queries I want to do

adam-urbanczyk, 2018-05-17:
My branch is based on pythonOCC actually. I am just saying that it is not
bundled with FreeCAD.

dcowden, 2018-05-17:
@adam-urbanczyk  oh ok i was confused, sorry!

How will we distribute pythonOCC along with a cq version that needs it?

gntech, 2018-05-17:
If we want to distribute cadquery through pypi and bundle pythonOCC and the OCC binaries with cadquery perhaps we do something similiar to the opencv-python package on pypi. It bundles prebuilt binaries of opencv in the pypi-package.
See their repo here: https://github.com/skvark/opencv-python

If we do it this way we can install via `pip install cadquery`

adam-urbanczyk, 2018-05-17:
@dcowden @gntech   for me the only workable option is conda; pip is not really up to the task of distributing non-python packages (i.e. oce ). I already have a working conda package for cq-occ ( https://anaconda.org/cadquery/cadquery-occ ).

dcowden, 2018-05-17:
Alright then! With no objections then let's move forward with integration of Adams branch.

zwn, 2018-05-18:
Will this mean that cadquery will not be usable from within freecad anymore?

Would it be possible to use the upstream version https://anaconda.org/pythonocc/pythonocc-core or does it have some problems? I see it has got builds for all the common platforms.

zwn, 2018-05-18:
Now I've found one more https://anaconda.org/cad/pyocct - it seems Freecad is moving towards using this package (see https://github.com/FreeCAD/FreeCAD_Conda).

dcowden, 2018-05-18:
@zwn we definitely have to make sure cq can still run in freecad  for some period of time, for cadquery freecad module. But I think the long term plan is to make our own gui, perhaps based on electron.

The perfect scenario would be if freecad actually used pythonocc, because then we could run in either environment with the same code. But i am afraid that won't happen. If it doesn't, we would have to support two codebases... One for cq on freecad and one for cq on OCC. This is not something I want to do long term. The reason for the move away from freecad is that there are tons of limitations and bugs we can't work around.

gntech, 2018-05-18:
I have installed the cadquery-occ package from https://anaconda.org/cadquery/cadquery-occ for some testing and found that the installation instructions was lacking a bit.

On that page it says install with `conda install -c cadquery cadquery-occ`

However, since there are some dependencies from other channels needed this command was not enough. The following command activates the necessary channels to get the dependencies and install cadquery-occ.

`conda install -c conda-forge -c DLR-SC -c oce -c pythonocc -c cadquery cadquery-occ`

@adam-urbanczyk If you agree, perhaps this could be added to the cadquery-occ installation instruction page?

adam-urbanczyk, 2018-05-18:
@gntech  Thanks for looking into this. I think that the line on anaconda.org is auto-generated, not sure how to change it.

Proper instruction can be found here: https://github.com/adam-urbanczyk/cadquery

fragmuffin, 2018-05-20:
:sleeping: :zzz: :bangbang:  :open_mouth: I'm awake!
Yep, I'm in!

I've created a ["OpenCascade (OCC)" milestone](https://github.com/fragmuffin/cqparts/milestone/2) for `cqparts`.
All changes will go in to the [`feature/freecad-occ`](https://github.com/fragmuffin/cqparts/tree/feature/freecad-occ) branch until it's ready to be merged & released.

@osterwood thanks for your good work.
If it's alright with you, I'll cherry-pick what you've done to give me a head start!

dcowden, 2018-05-21:
@here what do you think about this roadmap:

1. Create occ branch in dcowden/cq, starting with Adams version
2. Validate that tests pass
3. Set up ci loops on the branch to build conda packages on release. ( I like the way we do that on the main branch... A release is cut whether a tag is present)

4. Announce availability for those interested.

5. Make a docker image for those who want to run that way (I'll take this one)

6.  Start work on a cq gui. @jmwright, what are your thoughts about the best platform upon which to do that?

7. Run the branch in parallel with master until the library is solid and the new GUI works well

8. Merge non freecad cq in to master. Hopefully at this point freecad is using OCC , and it's possible to run cq inside of freecad, by adding pythonOCC only.

History teaches that cq was not of much interest until there was a gui. It really took off once Jeremy made cadquery freecad module.  Consequently,  I think we have to have a functional GUI before we can try dumping Freecad.  

The shortest path to a working GUI is probably to use the starting point from the pythonOCC samples, which in turn means using wx if I recall. 

You can see  A few at https://github.com/tpaviot/pythonocc-demos?

I can imagine building a gui starting from these pretty quickly.

zwn, 2018-05-21:
I see that `cadquery-occ` package is based on `pythonocc-core` (see https://github.com/adam-urbanczyk/cadquery/blob/cq1_pythonocc/conda/meta.yaml). 

Freecad is using `(py)occt` (see https://forum.freecadweb.org/viewtopic.php?t=12920) - according to the posts even 0.18 release of `oce` has some bugs already fixed in `occt 7.2`.

Is cadquery aiming at using `pyocct` or `pythonocc-core`?

dcowden, 2018-05-21:
@zwn good question. I didn't even know about pyocct until you brought it up.  In conversation, i've been talking about pythonOCC, but that's only because its the one i'm familiar with.  It's been around forever. 

As mentioned in the article you referenced, the most notable difference is that pythonOCC is based on OCE, not OCCT.  I'm not sure I care too much about that distinction. 

if freecad is moving to pyOCCT, that would be good news for freecad/cq interoperability, but bad news for @adam-urbanczyk 's branch, which is based on pythonocc.

i'd be interested in thoughts on the apis themselves.  pythonOCC started life as a simple set of auto-generated bindings, much like ( it appears ) pyOCCT is. However, over time, the pythonOCC guys have spent a lot of time tweaking the bindings to solve memory problems, garbage collection and such.  I wonder if pyOCCT is doomed to repeat that work?

zwn, 2018-05-21:
http://pyocct.readthedocs.io/en/latest/extra.html#pythonocc-comparison but I have no idea why there are two different bindings :-(. 

In pythonOCC there is also a qt based gui, see https://github.com/tpaviot/pythonocc-core/blob/master/src/Display/SimpleGui.py. I have not found anything similar in pyOCCT.

I'd just go with pythonOCC for now since it seems easier (a lot has been done already).

dcowden, 2018-05-21:
@zwn reading the front page of pyOCCT, this comment:

>  As new versions of OpenCASCADE are released, it may be better to patch up the existing bindings rather than rerun the binding generation tool. The automated process still requires a handful of manual patches to capture edge cases

Causes me to think they have a long path in front of them, and that it is very likely the same path pythonOCC has already walked. I think pythonOCC has a LOT of tweaks to their bindings, developed over literally a decade now. My hunch is that pythonOCC is the stronger play.

dcowden, 2018-05-21:
> I like using OCCT 7.2 as a basis, but I cannot find the binding generator in the PyOCCT codebase (maybe it is not shared on purpose; will open an issue).

I think you can't find it because it doesn't exist.  They say right on the front page that what's there is the result of manual work:

> The automated process still requires a handful of manual patches to capture edge cases

This causes me to worry about using it as a basis. In the long term, i like how they handle 'Handle'-- the code is much more intuitive.

adam-urbanczyk, 2018-05-21:
@dcowden  the way I read it is:
 there **is** an automated script, but the resulting code requires some manual edits.

Anyhow, let's see what the author will reply.

dcowden, 2018-05-21:
sounds good. regardless, i think the level of maturity has to be better with pythonOCC right?

zwn, 2018-05-21:
> PyOCCT is a quite new project that exposes OCCT 7.2 using state of the art pybind11 package (as opposed to SWIG and PythonOCC). To my understanding there is no relation of PyOCCT and FreeCAD.

I figured we can just ask: https://forum.freecadweb.org/viewtopic.php?f=10&t=28863

dcowden, 2018-05-21:
Good question!

zwn, 2018-05-21:
As for the GUI - could this be a start? 😉 

```python
from OCC.Display.SimpleGui import init_display

display,start_display, add_menu,add_functionto_menu = init_display()

import cadquery

result = cadquery.Workplane("front").box(2.0, 2.0, 0.5)

display.DisplayShape(result.toOCC())
start_display()
```

[image]

dcowden, 2018-05-21:
yep, exactly!!!

There are a few other features we would need (none of which are difficult):
   * add an editor pane so you can see the code you're editing and the rendered view
   * edit/save/load scripts
   * integrate with cqgi so that scripts do not have gui details in them

Regarding the first two, [CadQuery FreeCAD Module](https://github.com/jmwright/cadquery-freecad-module) is a pretty good reference for what the 1.0 version of the gui would need to do: people are generally pretty happy with it.

This last one bears some discussion.  cqgi's goal is to allow scripts to be portable between execution environments. More specifically, we want the scripts listed in the cq examples to run without modification in the cq gui, and in cadquery freecad module.

 [cqgi](https://dcowden.github.io/cadquery/cqgi.html) is an interface designed to allow scripts to be run in any gui, without having code specific to that gui ( for example, start_display() and all that makes this script not portable).  

[this example](https://github.com/dcowden/cadquery/blob/master/examples/FreeCAD/Ex001_Simple_Block.py) should be executable in the new GUI without need for changes.  There is currently a discussion about it going on in [cqparts](https://github.com/fragmuffin/cqparts/issues/66#issuecomment-388377281).  I think we've agreed in that discussion that we should add an explicit 
`from cadquery import *
 from cadquery.cqgi import show_object
` 
at the top of the script-- @fragmuffin believes that the fact that the show_object method appears defined out of nowhere is confusing, and i agree with him.  

So a cqgi- compliant version of your example would read:
```
import cadquery
from cadquery.cqgi import show_object
result = cadquery.Workplane("front").box(2.0, 2.0, 0.5)
show_object(result, {
    'color': 'yellow',
    'fit': 'extents'
})

```
Other executing environments ( like the autogenerated spinx doc ) should ignore what they can't handle. 
cqgi's debug method is designed to highlight the objects in red when you're having trouble with a script. For example, if you do:

```
from cadquery.cqgi import debug_object,show_object
c = Workplane('XY').circle().extrude().faces(">Z")
debug_object(c)
..
show_object(result)

```
You should see the cylinder rendered, and the top face should be highlighted red.

Other future features cqgi should provide would be profiling ( how long did various things take to run), and parametrics ( detecting parametric variables in a script and allowing the user to change those values without changing the source code).

adam-urbanczyk, 2018-05-21:
@dcowden @zwn 

FYI: I am refactoring the "old" experimental UI based on PyQT and pyface [I made the repo private to reduce noise for now].

Design goals/choices :
* PyQt5/python3 only
* Do not use other frameworks (i.e. get rid of pyface, traits)
* Reuse the editor widget from Spyder IDE
* Add some debugging functionality 

I will try to make it public within ~4 wks

[image]

jmwright, 2018-05-21:
I built an early experimental Qt GUI called [cadquery-x](https://github.com/jmwright/cadquery-x), and I think @adam-urbanczyk built a more complete variation at some point. I'm not sure if cadquery-x will even run anymore though without a bunch of updating. I also created [cqcad](https://github.com/jmwright/cqcad) which is a Qt GUI shell without everything implemented yet (it does have CI integration though). I was working on it but there was more interest in the Electron-based GUI [cadquery-gui](https://github.com/jmwright/cadquery-gui), so I switched back to it. There's also the option to update the FreeCAD module to work with CQ 2.x, and I will probably do that at some point after the 2.x branch is stable.

As far as GUIs, I don't know what the answer is. There's too much variation in what users want. @fragmuffin and @RustyVermeer seem to favor a solution which opens a rendering window and watches the script being edited. That allows CQ users to use their favorite editor/IDE on the scripts, and the 3D view updates automatically. It's pretty nice for users who grok that kind of workflow, and is probably the most powerful and flexible for a large range of users. However, it isn't a one-size-fits-all solution. Other users (myself included sometimes) use FreeCAD and CadQuery as tools in a toolchain, so something like CQFM (the FreeCAD module) works best. And beyond that some users will want the full package stand-alone GUI with batteries included, like cqcad (or what it's designed to be).

The quickest to implement with the best user coverage would probably be to include a watching renderer with CadQuery 2.0 installs (Docker, Anaconda, whatever). This is the approach where you run a renderer app and pass it a file name, and it renders that (web or otherwise) and then watches the file for changes. Starting out, most CQ 2.0 users are going to be early adopters and will probably be willing to deal with not having one integrated environment to edit scripts in.

jmwright, 2018-05-21:
I think @adam-urbanczyk and I posted at about the same time. +1 on his GUI. It looks like a more complete version of where I was headed with cqcad.

dcowden, 2018-05-21:
ok so @adam-urbanczyk, UDMAN!  I think we plan on merging your branch into the main cq as a new version, with the idea that its the first complete stand alone version.  

on that branch, we'll change cqgi per some of the comments by @fragmuffin , so that we're all happy with how the scripts interact with the gui.  

We'll also update documentation on installation, and set up the CI to make conda packages.

Then we'll announce a conda-one step install of cq! exciting!

zwn, 2018-05-21:
> thoughts?

I'd go with anything that has the minimal amount of dependencies and is the least amount of work. Freecad looks like a dependency nightmare where several people need to work just on packaging for different platforms. Also the scope is too big so nothing really works all the way without glitches.

Therefore I don't see a builtin editor as a must-have (lots of editors have been built already, why build another one with limited manpower that could have a better use). So watching a .py file and rebuilding on changes is ok with me. 

OTOH "conda one step install" is a must-have for me (for windows and linux). It would be nice if the dependencies were in channel "defaults" or "conda-forge" so someone else does the maintenance 😉.

dcowden, 2018-05-21:
good thoughts @zwn ! I agree completely with the one-step install/conda package, its a must. I think its also not a lot of work since @adam-urbanczyk already has a conda package building.  I'm not sure if all the deps he used were in defaults or conda-forge

zwn, 2018-05-21:
@dcowden I've taken look at cqgi, let me know if this is the proper way to do it:

```python
from cadquery import cqgi

parsed = cqgi.parse(
"""
obj = cadquery.Workplane("front").box(2.0, 2.0, 0.5)
build_object(obj)
"""
)
built = parsed.build()

from OCC.Display.SimpleGui import init_display

display,start_display, add_menu,add_functionto_menu = init_display()

display.DisplayShape(built.results[0].toOCC())
display.FitAll()
start_display()
```
There are for sure lots of details to be handled but for lack of better options right now I'll stay with this.

dcowden, 2018-05-21:
yep, that's basically it!  if that constant string were loaded from a file or from the users' text editor, then the script doesnt have that extra stuff.

Two modifications: 

(1) build_object should be show_object-- that was changed but the documentation is wrong i guess :(

(2) Per comments from @fragmuffin  and some others, I think we've agreed to require the script to import cadquery and build_object function. In the current implementation, i was trying to keep the script as simple as possible-- but I ran afoul of common sense, because leaving those lines out is confusing-- every wanted to know where these magic methods came from, and what other ones are in scope.

So though its not implemented yet, i think it would be:

```
from cadquery import cqgi
parsed = cqgi.parse(
"""
import caduery
from cadquery.cqgi import show_object
obj = cadquery.Workplane("front").box(2.0, 2.0, 0.5)
show_object(obj)
"""
)
built = parsed.build()

from OCC.Display.SimpleGui import init_display

display,start_display, add_menu,add_functionto_menu = init_display()

display.DisplayShape(built.results[0].toOCC())
display.FitAll()
start_display()
```

when we're operating in a pythonOCC based cq version, we could lose the toOCC because we could assume all objects are OCC objects.

zwn, 2018-05-21:
@dcowden What's next? Who is doing what? 😉

dcowden, 2018-05-21:
@adam-urbanczyk has a gui he's willing to clean up-- estimated 4 wks for delivery. once he's got that ready, i'll make a branch and pull it in, and we can iterate from there.

between now and then, i need to change cqgi to require the explict imports, per @fragmuffin -- i'll try to get to that this week

zwn, 2018-05-21:
Adams branch is based on 1.0.0 and that is why I used `build_object` instead of `show_object` - to get something that works today, right now. Were there a lot of changes between 1.0.0 and 1.2.0?

dcowden, 2018-05-21:
there were not a lot of changes, but the change from build_object to show_object was in that release. ( All of the [examples](https://github.com/dcowden/cadquery/tree/master/examples/FreeCAD) use show_object as of 1.2 (now) )

rowanG077, 2018-05-22:
I'm not sure how far the UI is along but I would suggest not creating a UI ourselves. I would rather use Visual Studio Code as the base and just create UI in that. And only provide a basic view layer within the framework itself.

jmwright, 2018-05-22:
@rowanG077 It seems like the UI @adam-urbanczyk is pretty far along. There was also talk at one time about using Atom as the base editor and adding the view layer to that.

zwn, 2018-05-25:
The default conda package for `pythonocc-core` brings in `pyqt` that is distributed under the GPL 😟 . I've created an issue https://github.com/tpaviot/pythonocc-core/issues/538 to see if this could be changed. Otherwise there is a little point of distributing `cadquery` under Apache license.

adam-urbanczyk, 2018-05-25:
> tpaviot/pythonocc-core#538

How is the pyqt license a problem for us? The PythonOCC conda package is
definitely not a compiled executable, it doesn't even contain the pyqt code.

zwn, 2018-05-25:
While I am no licensing expert, I am reading that PySide has been created for the sole reason that the original copyright holder would not allow distribution under LGPL. So while I don't fully understand the situation, I trust that this decision has not been taken lightly. Also please note that pyqt is full of binary/compiled extensions. AFAIU when you load a dynamic library distributed under GPL you are bound to distribute the rest of the application under GPL as well. I'd be happy to be proven wrong but in the meantime I'd rather be on the safe side.

For more info please see:
* https://en.wikipedia.org/wiki/PyQt
* https://riverbankcomputing.com/commercial/license-faq
* https://opensource.stackexchange.com/questions/6033/can-a-non-gpl-python-program-use-gpl-python-module
* https://www.gnu.org/licenses/gpl-faq.en.html#MereAggregation

zwn, 2018-05-25:
> The PythonOCC conda package is definitely not a compiled executable, it doesn't even contain the pyqt code.

PythonOCC conda has a dependency on pyqt package which contains many compiled executables.

zwn, 2018-05-25:
https://softwareengineering.stackexchange.com/questions/87446/using-a-gplv3-python-module-will-my-entire-project-have-to-be-gplv3-licensed

The accepted answer contains a citation from GPL that seems to make look it very clear.

adam-urbanczyk, 2018-05-25:
We are distributing code, I don't get your point.

   - https://en.wikipedia.org/wiki/PyQt
   - https://riverbankcomputing.com/commercial/license-faq
   -
   https://opensource.stackexchange.com/questions/6033/can-a-non-gpl-python-program-use-gpl-python-module
   - https://www.gnu.org/licenses/gpl-faq.en.html#MereAggregation

Peque, 2018-05-26:
I am not a lawyer, so do not take my opinion as any kind of serious advice.

I agree with @zwn and I think an Apache project should not make use of any GPL library, specially if they may contain compiled code. As [stated here](https://www.apache.org/licenses/GPL-compatibility.html) Apache and GPL are compatible in one way only.

I once created a non-free-software project that required Qt Python bindings and the only reason why I chose PySide was the LGPL license. Also, as of today, there is PySide2 which fortunately is backed up by Qt as the official Python bindings.

Anyway being CadQuery a non-commercial project I do not think you will ever have any lawsuits or legal issues. Also, if PyQt is a dependency added by `pythonocc-core` and not CadQuery itself it could be less of a problem, as `pythonocc-core` is LGPL licensed.

Maybe the safest path would be to modify CadQuery's license, if you are willing to do so. Otherwise the safest path would probably be to make sure there are no GPL-licensed libraries used within the project.

dcowden, 2018-05-26:
We discussed licensing at length in the early days.  cadquery is useful to
people in the scientific and corporate world-- we wanted to stay away from
GPL so that these uses are not constrained.

zwn, 2018-05-26:
> We are distributing code, I don't get your point.

While the license in the repo says "apache", if someone uses the conda package of cadquery-occ, it is my understanding they are actually getting "gpl" without knowing. The original cadquery license allows to build closed source applications while the gpl does not.

dcowden, 2018-05-26:
What if we distribute the gui in a separate package?

adam-urbanczyk, 2018-05-26:
Precisely, it is up to the user to get a commercial pyqt license (not that
expensive actually) if he/she wants to redistribute the SW further without
sharing the code as a whole under gpl. I don't see any problem with that.

fragmuffin, 2018-10-27:
Hi all, I'm a bit behind with all of this.
I'm trying to find some more time outside work to get back into open source.

**Question:** what branch is the focal point for OCC testing and updates?

dcowden, 2018-10-27:
I would like to use https://github.com/dcowden/cadquery/tree/python_occ_2_0?files=1

For this work

jmwright, 2018-10-28:
At one point I thought we had decided to start a new cadquery repo under the CadQuery organization and base that on the work that's been done on the fork @adam-urbanczyk has. Then eventually sunset this CQ 1.0 repo.

zignig, 2018-10-28:
The discussion is here 

https://groups.google.com/forum/#!topic/cadquery/ZbjCNAszLnA

I think that it is a good idea to move cadquery into the org, and even (@fragmuffin) move cqparts into it's own org.

From there the conversion to an OCC base can be done in isolation.

dcowden, 2018-10-28:
Right! I remember that conversation now.
We decided to leave the exiting cq alone and the make the new cq in the cadquery org be the occ based one. We would use master there

Is everyone ok with that? What code should we start with for that? I thibk Adam's occ branch is probably the best?

jmwright, 2018-10-28:
I think Adam's is the most complete and up to date.

If everyone agrees, I guess the next step would be to create a cadquery repo on the org, then for Adam to rebase/merge/whatever his fork/branch, change the remote setup, and push to the new repo. My git knowledge is lacking compared to some members of the community though, so there may be better ways to do it.

adam-urbanczyk, 2018-10-28:
+1 @jmwright

[28 more replies]