dcowden/cadquery #269: Implement subtractive features

adam-urbanczyk, 2018-05-21:
For now only additive features are implemented. If we need to make a subtractive feature two additional calls are needed (combine=False and an explicit cut op)

```python
obj = cq.Workplane('XY').circle(dia_out).circle(dia_in).extrude(tube_h)
groove= obj.faces('<Z').workplane(offset=-3.0).circle(dia_out).\
    circle(dia_out-groove_depth).extrude(groove_h,combine=False)

result = obj.cut(groove)
```

What would be nice:
```python
result = cq.Workplane('XY').circle(dia_out).circle(dia_in).extrude(tube_h).\
    faces('<Z').workplane(offset=-3.0).circle(dia_out).\
    circle(dia_out-groove_depth).extrude(groove_h,subtract=True)
```

dcowden, 2018-05-21:
cutBlind, cutThruAll perform cuts using a sketch or wires, and avoids the extra step.

dcowden, 2018-05-29:
I definitely wouldn't mind, wouldnt hurt anything.

My main reservation is that we've come 'under fire' recently for the use of kw args-- novice users complain that they are hard to figure out and generally not obvious that they are there.   IE, i think novice users would most likely never know it's an option.

That's not an argument against the new argument, as much as that we generally have to figure out how to make the functions more self-describing. But i have no idea how to do that, because a lot of our operations do have lots of choices