dcowden/cadquery #270: CQGI: require explicit cadquery and show_object imports

dcowden, 2018-05-21:
after discussions with @fragmuffin and others, we'd like to change cqgi to require scripts to import show_object, debug, and cadquery itself.

The consensus was that there is just too much magic in scripts that use show_object when it appears out of nowhere.

This will require changing cqgi itself, as well as changing all of the examples to do this import. It probably also means that cqgi.show_object should be promoted to a real function, not injected during the ast phase.

@jmwright before i start these changes, do you agree? What is the impact on cqfm?

dcowden, 2018-05-21:
created based on conversation in #281 and also in https://github.com/jmwright/cadquery-freecad-module/issues/112, and

dcowden, 2018-05-21:
and also mentinoed in https://github.com/fragmuffin/cqparts/issues/66#issuecomment-390788722

jmwright, 2018-05-22:
> @jmwright before i start these changes, do you agree? What is the impact on cqfm?

There will be some changes. CQFM assumes that show_object magically shows up.

Probably the biggest challenge is that this change will break existing scripts. When people upgrade CQ their scripts are going to break and they won't understand why.

zwn, 2018-05-22:
Having the possibility to import `cadquery` and `show_object` would open the possibility to execute the script as is. Eventually, `show_object` can be implemented using pythonOCC SimpleDisplay. That would allow (I think) easy onboarding with no magic involved. I think that only cq developers should have the need to know about cqgi. Users of cq could stay oblivious and I think that is a good thing. Tutorial on "getting started with cq" would involve conda install plus running the examples as regular scripts.

rowanG077, 2018-05-22:
What is the timetable for 2.0? Maybe include this change in the 2.0 version and break backwards compat. I rather have a clean interface and change a couple of scripts then have a murky interface that is confusing to newcomers.

dcowden, 2018-05-22:
The timetable for cq 2.0 indeterminate at this point.
I think we are all in agreement that 2.0 is where we introduce breaking changes.

rowanG077, 2018-05-22:
Yeah so it would make more sense to make this a 2.0 feature.

fragmuffin, 2018-05-25:
> I need to figure out a way to make the changes reverse compatible.

I don't think that's possible, if the target is to "remove magic" (which I totally agree with 👍 ), then old scripts will error with: `NameError: name 'show_object' is not defined`

To ease the transition, it may be worth catching that exception at a higher level, and print a message...

```python
try:
    run_script() #?
except NameError as e:
    if e.message == "name 'show_object' is not defined":
        print("show_object method not supported by cadquery v2.0, see https://goo.gl/JsWyUP")
    raise
```

Although, again, this is python running from python... so I'm not sure where the magic ends.

closed 2018-12-06 by dcowden