dcowden/cadquery #274: CadQuery team logo

fragmuffin, 2018-05-26:
Hey @adam-urbanczyk , could we get the [`cq` logo](https://github.com/dcowden/cadquery/blob/master/doc/_static/logo/cadquery_logo_dark.svg) as he profile pic on the [CadQuery team page](https://github.com/cadquery)?

I know it's a small thing, but it's been bugging me, and I can't do it myself :smile:

adam-urbanczyk, 2018-05-29:
Done

jmwright, 2018-05-29:
Thanks @adam-urbanczyk 

What's the plan with the team page? Are we ever going to transfer repos to it at some point?

@dcowden

dcowden, 2018-05-29:
@adam-urbanczyk @jmwright i have no idea what github teams mean and how they function.  Is this something you guys would like to do? How does it help the community?  I'm <slightly> worried about re-locating the repository due to links to the old path-- but of course we could leave a re-located link.

closed 2018-06-07 by fragmuffin