dcowden/cadquery #279: CQGI Does Not Work With cqparts Assemblies

jmwright, 2018-06-06:
I can't seem to get CQGI to accept a cqparts.Assembly script.

I can use `show_object` with a cqparts part script which includes raw CadQuery code. However, when I try to use parts in an assembly, which is an abstraction level above raw CQ code, I get the error `name 'show_object' is not defined`. I suspect that CQGI is enforcing that the script includes raw CadQuery code somehow. Below is my script.
```python
import cadquery as cq
import cqparts
from cqparts.params import *
# from cqparts.display import display
from cqparts.utils import CoordSystem
from cqparts.constraint import Fixed, Coincident
from parts.chassis.chassis import Chassis
from parts.cots.futaba_s3003_servo import Servo

class RobotAssembly(cqparts.Assembly):
    def make_components(self):
        chassis = Chassis()

        return {'chassis': chassis}

    def make_constraints(self):
        return [
            Fixed(self.components['chassis'].mate_origin, CoordSystem())
        ]

robot_assembly = RobotAssembly()
show_object(robot_assembly.local_obj, options={"rgba": (204, 204, 204, 0.0)})
```
@dcowden Any ideas on a work-around for this? @fragmuffin Have you seen this problem?

dcowden, 2018-06-06:
@jmwright i did a little testing on this. Here's an example session that shows typical use ( run from within cadquery directory):

```
dcowden@dcowden-HP-Pavilion-dm4-Notebook-PC:~/gitwork/cadquery$ python
Python 2.7.12 (default, Dec  4 2017, 14:50:18) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import cadquery
>>> import cadquery.cqgi
>>> example_script = open('./examples/FreeCAD/Ex001_Simple_Block.py').read()
>>> block_model = cadquery.cqgi.parse(example_script)
>>> block_model.metadata.parameters
{'length': <cadquery.cqgi.InputParameter instance at 0x7fbb1586c5f0>, 'thickness': <cadquery.cqgi.InputParameter instance at 0x7fbb1581c2d8>, 'height': <cadquery.cqgi.InputParameter instance at 0x7fbb1581c290>}
>>> block_result = block_model.build()
>>> block_result.success
True
>>> dir(block_result)
['__class__', '__delattr__', '__dict__', '__doc__', '__format__', '__getattribute__', '__hash__', '__init__', '__module__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', 'buildTime', 'debugObjects', 'exception', 'first_result', 'results', 'set_debug', 'set_failure_result', 'set_success_result', 'success']
>>> block-result.results
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'block' is not defined
>>> block_result.results
[<cadquery.cqgi.ShapeResult object at 0x7fbb15823690>]
```

CQGI works by injecting the show_object method into the top level script.  My guess is that cqparts doesnt work right because something is calling show_object from within a scope other than the top level scope.    More specifically, i would guess that somewhere inside of the call to Chassis(), show_object is called. Since show_object is only injected into the top level script, it is not defined in the subscope.

jmwright, 2018-06-06:
@dcowden Thanks, that must be the problem. That puts me in a bit of a bind though. I have my files stored on STEMN, and have the pipeline generating STLs of some of the parts. If I pull the show_object function out of those, the won't export properly anymore.

Will a resolution to #270 fix this as well?

dcowden, 2018-06-07:
Yes, it would, and it is probably the right solution for general scripts, which would include examples, as well as scripts a that are kicking off cqparts builds.

It's probably not the best solution for native cqparts though.. @fragmuffin and I are talking about an adapter that would be better for that purpose

jmwright, 2018-06-07:
I can solve this problem for now on STEMN using environment variables. This issue kind of seems like a duplicate of a couple of others, so I'm going to close it.

closed 2018-06-07 by jmwright