dcowden/cadquery #291: RFC: Polar line and lineTo methods

johnbeard, 2018-08-20:
I use the following methods for defining lines in terms of polar co-ordinates, which I sometimes find helpful (`hLine`/`vLine` are useful, but not all lines are on the axes). I don't know if there is appetite to add this to the base `Workplane` object, but I thought I'd post it just in case!

I haven't tried yet to set up a CQ development environment, but I'm happy to give that a go and make a PR if helpful.

``
def polarLine(self, distance, angle, forConstruction=False):
    """
    Make a line of the given length, at the given angle from the current point
    
    :param float diastance: distance of the end of the line from the current point
    :param float angle: angle of the vector to the end of the line with the x-axis
    :return: the Workplane object with the current point at the end of the new line
   """
    x = math.cos(math.radians(angle)) * distance 
    y = math.sin(math.radians(angle)) * distance

    return self.line(x, y, forConstruction)
    

def polarLineTo(self, distance, angle, forConstruction=False):
    """
    Make a line from the current point to the given polar co-ordinates
    
    Useful if it is more convenient to specify the end location rather than
    the distance and angle from the current point
    
    :param float diastance: distance of the end of the line from the origin
    :param float angle: angle of the vector to the end of the line with the x-axis
    :return: the Workplane object with the current point at the end of the new line
    """
    x = math.cos(math.radians(angle)) * distance 
    y = math.sin(math.radians(angle)) * distance
    
    return self.lineTo(x, y, forConstruction)

#link the plugin into cadQuery
cq.Workplane.polarLine = polarLine
cq.Workplane.polarLineTo = polarLineTo
``

Thanks for CQ, and thanks for making extensions so easy!

jmwright, 2018-08-20:
@johnbeard +1 on this. It's basically what I had in mind when I created issue #75 

If you would like to get set up to do a PR, that would be great. We can offer support as you work through it.

johnbeard, 2018-08-20:
I guess my main real question is: how do you run FreeCad with a development copy of CadQuery rather than the "installed" one? Or should I uninstall and have two versions - one "stable" and one "dev" and choose using the `-M` parameter at launch?

jmwright, 2018-08-20:
@johnbeard The way I do it is to install our FreeCAD module, and then replace the [cadquery lib directory](https://github.com/jmwright/cadquery-freecad-module/tree/master/Libs/cadquery) with a cloned version of my fork. It oftentimes takes a restart of FreeCAD to pick up the changes, but it makes it farily easy to develop. You can write scripts to automate a lot of that too. I've had scripts that replaced the cadquery directory in the module with an externally cloned one, then copied the module into the system's FreeCAD Mod directory, then launched FreeCAD. There's quite a bit of latitude there because the cadquery library is just a directory. Swap it out with your development copy, restart FreeCAD, and you're on your development version of the library.

One word of caution though. Sometimes when you clone a git repo into another one directly, the git cache/tree/whatever can do things that are unexpected (at least in my experience). I'm sure whatever git does is perfectly logical, I've just had some counter-intuitive things happen before.

johnbeard, 2018-08-22:
Thanks, I used that info to make a handy technique:

* Uninstall `cadquery_module`
* Clone `cadquery` to a directory (say `~/src/cq/cadquery`). This is the development copy.
* Clone `cadquery_module` to a directory (say `~/src/cq/cadquery_module_original`)
* Clone `cadquery_module` again to another dir (say `~/src/cq/cadquery_module_dev`)
* In the `_dev` version, change InitGui.py to read "CadQuery dev" so it's obvious what you're running
* In the dev version, delete `Libs/cadquery` and replace with a symlink to your `cadquery`

Now, you can run one or the other:
* `freecad -M ~/src/cq/cadquery_module_dev`
* `freecad -M ~/src/cq/cadquery_module_original`

You can then set up a `.desktop` file or shell alias for each so you can launch quickly.

Development of cadquery is then done in `~/src/cq/cadquery`, so Git only sees a single tree and it's all "normal".

It seems as you say you must reload FreeCAD to pick up new changes in CQ files.

N.B. You can't trivially run them both, even if you rename `CadQueryWorkbench` as the import will only pick up one copy of the library, although you *can* get a menu entry for each.

jmwright, 2018-08-26:
@johnbeard Very cool technique. I think I'll adopt that for my development setup too.

jmwright, 2018-08-26:
Closing this now that #292 had been merged. Feel free to reopen for the other features that were talked about in the PR.

closed 2018-08-26 by jmwright