dcowden/cadquery #293: How did you manage the security concerns?

ceremcem, 2018-09-06:
Hi,

I'm currently using FreeCAD and I needed some scripting support where the script should be distributed by the document and executed upon document open/update. We have discussed the issue [here](https://forum.freecadweb.org/viewtopic.php?f=3&t=30731) and it seems the only problem is the security concerns. 

If running any code would be that dangerous, how did you managed to stay in the safe side? I think the same precautions might be ported into FreeCAD.

jmwright, 2018-09-12:
Our FreeCAD workbench doesn't autorun scripts when you open them unless you override the setting for that. We rely on the user to ensure that the script they're running is safe before executing it. I'm not aware of any in-built security checks, although our CQFM implementation kind of sandboxes scripts. However, I doubt it's enough to avoid security concerns due to executing malicious code.

ceremcem, 2018-09-12:
> We rely on the user to ensure that the script they're running is safe before executing it

I think that can't actually serve as a precaution because we can't always read the models we get. We could verify at the beginning and we'll quit when it gets a burden some time later. It will be like [Windows popups](https://www.google.com.tr/search?safe=active&ei=g0mZW7yONsvRrgTdkIK4Bg&q=windows+administrator+rights+popup&oq=windows+administrator+rights+popup&gs_l=psy-ab.3...17698.18606.0.19171.7.7.0.0.0.0.216.762.0j2j2.4.0....0...1.1.64.psy-ab..3.1.215...0i30k1j0i8i30k1.0.1ivn2JhMJBk), I mean, people will just run the code without reading. 

> CQFM implementation kind of sandboxes scripts

That is what I hope to hear in the first place. I *think* that this sandboxing approach has to be perfectly enough to stay in the safe side. I mean, how does [Repl.it](https://repl.it/@whearst3/Python-Sandbox) manage to do this goal?