dcowden/cadquery #298: Segfault while trying to create fillet

FrederikLauber, 2018-11-21:
Hello,

trying to soften the lower edge of one of my projects but am always greeted with a segfault then trying to do this. I attached the project as well as 2 videos of the crash.
Sometimes freecad just freezes then adding all the facets which are supposed to become round. 
Cheers,

Technical Infos:
Running Arch Linux with FreeCAD 0.17 Revision: 13543 (Git)

This is the error message:
** (process:12949): WARNING **: 21:21:31.099: XDG_TEMPLATES_DIR is set to invalid path, ignoring it
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x37e00) [0x7ff296267e00]
#1  0x7ff2689832d1 in ChFi3d_FilBuilder::PerformTwoCorner(int) from /usr/lib/libTKFillet.so.7+0x6701
#2  0x7ff2688d6152 in ChFi3d_Builder::PerformFilletOnVertex(int) from /usr/lib/libTKFillet.so.7+0x3b2
#3  0x7ff2688d769f in ChFi3d_Builder::Compute() from /usr/lib/libTKFillet.so.7+0xe3f
#4  0x7ff268a4f6e4 in BRepFilletAPI_MakeFillet::Build() from /usr/lib/libTKFillet.so.7+0x34
#5  0x7ff25f4140b9 in PartDesign::Fillet::execute() from /usr/lib/freecad/lib/_PartDesign.so+0x4f9
#6  0x7ff2978b95fd in App::DocumentObject::recompute() from /usr/lib/freecad/lib/libFreeCADApp.so+0x7d
#7  0x7ff268c92b6c in Part::Feature::recompute() from /usr/lib/freecad/lib/Part.so+0xc
#8  0x7ff297886085 in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x55
#9  0x7ff29683aa7c in QMetaObject::activate(QObject*, int, int, void**) from /usr/lib/libQt5Core.so.5+0x77c
#10  0x7ff29808dfd4 in Gui::QuantitySpinBox::valueChanged(double) from /usr/lib/freecad/lib/libFreeCADGui.so+0x44
#11  0x7ff29808f0df in Gui::QuantitySpinBox::userInput(QString const&) from /usr/lib/freecad/lib/libFreeCADGui.so+0x51f                               
#12  /usr/lib/freecad/lib/libFreeCADGui.so(+0x649662) [0x7ff298092662]                                                                                
#13  0x7ff29683aa7c in QMetaObject::activate(QObject*, int, int, void**) from /usr/lib/libQt5Core.so.5+0x77c                                          
#14  0x7ff296d31b63 in QLineEdit::textChanged(QString const&) from /usr/lib/libQt5Widgets.so.5+0x33                                                   
#15  /usr/lib/libQt5Widgets.so.5(+0x29771a) [0x7ff296d3871a]                                                                                          
#16  0x7ff29683aa7c in QMetaObject::activate(QObject*, int, int, void**) from /usr/lib/libQt5Core.so.5+0x77c                                          
#17  0x7ff296d392f6 in QWidgetLineControl::textChanged(QString const&) from /usr/lib/libQt5Widgets.so.5+0x36                                          
#18  0x7ff296d3c760 in QWidgetLineControl::finishChange(int, bool, bool) from /usr/lib/libQt5Widgets.so.5+0x290                                       
#19  0x7ff296d3ca97 in QWidgetLineControl::internalSetText(QString const&, int, bool) from /usr/lib/libQt5Widgets.so.5+0x1a7                          
#20  0x7ff296d33af7 in QLineEdit::setText(QString const&) from /usr/lib/libQt5Widgets.so.5+0x67                                                       
#21  0x7ff2980902ed in Gui::QuantitySpinBox::stepBy(int) from /usr/lib/freecad/lib/libFreeCADGui.so+0xcd                                              
#22  0x7ff29683b30b in QObject::event(QEvent*) from /usr/lib/libQt5Core.so.5+0x9b                                                                     
#23  0x7ff296c37b0c in QWidget::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0x68c                                                                 
#24  0x7ff296d945aa in QAbstractSpinBox::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0x5a                                                         
#25  0x7ff298093e47 in Gui::QuantitySpinBox::event(QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x37                                           
#26  0x7ff296bf7e14 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x84                                    
#27  0x7ff296bff6e1 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x211                                                 
#28  0x7ff297dd9e69 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x89                                 
#29  0x7ff296810c39 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x179                                       
#30  0x7ff29686339a in QTimerInfoList::activateTimers() from /usr/lib/libQt5Core.so.5+0x42a                                                           
#31  /usr/lib/libQt5Core.so.5(+0x2c4c22) [0x7ff296863c22]                                                                                             
#32  /usr/lib/libglib-2.0.so.0(g_main_context_dispatch+0x2bf) [0x7ff2937863cf]                                                                        
#33  /usr/lib/libglib-2.0.so.0(+0x6cf89) [0x7ff293787f89]                                                                                             
#34  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7ff293787fce]                                                                        
#35  0x7ff296863fc9 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x69                  
#36  /usr/lib/libQt5XcbQpa.so.5(+0xd7da2) [0x7ff28d71bda2]                                                                                            
#37  0x7ff29680f8cc in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x13c                                    
#38  0x7ff296817bc6 in QCoreApplication::exec() from /usr/lib/libQt5Core.so.5+0x96                                                                    
#39  0x7ff297d93112 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0x1282                                           
#40  freecad(main+0x712) [0x558cb755d672]                                                                                                             
#41  /usr/lib/libc.so.6(__libc_start_main+0xf3) [0x7ff296254223]                                                                                      
#42  freecad(_start+0x2e) [0x558cb755d9ae]
          
[project_and_video_of_crash.zip](https://github.com/dcowden/cadquery/files/2605723/project_and_video_of_crash.zip)

jmwright, 2018-11-21:
@FrederikLauber Can you post your CadQuery script's .py file too?

FrederikLauber, 2018-11-21:
Hi,

I did the drawing by hand so I am unsure how one would get a CadQuery script out of the model...
I have to say, CadQuery looks like a pretty neat idea for a future project.

jmwright, 2018-11-22:
@FrederikLauber I'm a little confused. You don't seem to be using CadQuery anywhere in the videos. You're in the Part Design workbench, which is maintained by the FreeCAD team. Am I missing something?

FrederikLauber, 2018-11-22:
Oh yeah, just realized this is the CadQuery issue tracker not the freecad one. m(
Sorry to have bothered you guys with unrelated stuff, need to go now and hide below a rock somewhere.....

closed 2018-11-22 by FrederikLauber

fragmuffin, 2018-11-22:
@FrederikLauber the part you're building _does_ look like a good candidate for a `cadquery` script.
If you don't mind me asking: what is it for?

jmwright, 2018-11-22:
I don't get segfaults with FreeCAD 0.17 as much as with earlier versions, but fillleting complex profiles will still do it sometimes. Hopefully they can help you on the FreeCAD forums.

FrederikLauber, 2018-11-22:
Its a holder for an display of a half assembled IceCube DOM. Basically a 3D printed part that connects a Photomultiplier tube with a bunch of electronics. If you take a look at this picture https://southpoledoc.files.wordpress.com/2012/07/5-dom-picture-1588691480.jpg its the dark orange part at the neck (upper, thinner part) of the Photomultiplier (labeled as PMT here)

jmwright, 2018-11-22:
@FrederikLauber @fragmuffin Is right that it wouldn't be too hard to create that in CadQuery.

On your 3D model, there are lines going through the middle of the small circles on the perimeter of the ring that I don't think should be there. There's also a line going down the side of the ring that seems to bisect the ring. All of those are a little concerning for the fillet operation.

Have you tried doing a union boolean operation to make sure all the solids are combined before trying the fillet operation? Did you extrude one continuous profile to make that part, or did you create the ring and add the smaller circles later? From the looks of it I would say the second one, but it's hard to be sure.

fragmuffin, 2018-11-22:
oh wow! that's so interesting!
Main article: https://southpoledoc.wordpress.com/2012/07/09/icecube-neutrino-observatory/

Are you considering using an automated, or parametric approach for designing these?
If so, have a look at the [`cadquery` documentation](http://dcowden.github.io/cadquery/), and if you're still interested, read on with [`cqparts` documentation](https://cqparts.github.io/cqparts) (that's my baby)

I'm sure you'd get help from the `cadquery` community if you'd like help getting started.

FrederikLauber, 2018-11-22:
@jmwright Second one. I did not find a way to add a polar operation to a sketch in freecad. So I extruded first and then did the polar operation on top. The parts should be connected though as the borders are constructed exactly the same (big circle on the one side, segment of a circle with 3 points on the big circle). I haven't tried to do the Boolean operation yet. Manually modifying my 3D print with a file worked though ;) 

@fragmuffin Printing it without the fillet works fine and as I only needed one I added them manually in with a file. Everything else from here is just learning opportunity. I will have a look at cadquery and consider it for the next project. The idea of it looks very interesting, have to try it in the future!

easyw, 2018-11-22:
@FrederikLauber 
you may consider to use FC daily within OCC 7.3 ... there are many improvements in OCC and with this release I can fillet your model within a 2mm fillet.
[image]

I did it in Part WB and not in Part Design WB... it seems it works better for fillets ...
[DOM_HOLDER_V6.zip](https://github.com/dcowden/cadquery/files/2607686/DOM_HOLDER_V6.zip)

PS nice to see FreeCAD used for these interesting projects 😄

fragmuffin, 2018-11-22:
@FrederikLauber I'm glad you got what you needed.
Let us know if you have any questions, `cadquery` and `cqparts` has an awesome community of friendly, helpful, and clever people.

@easyw I was also able to apply a 2mm fillet using `v0.17` (details below).

```
OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13528 (Git)
Build type: Release
Branch: releases/FreeCAD-0-17
Hash: 5c3f7bf8ec51e2c7187789f7edba71a7aa82a88b
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/Australia (en_AU)
```