dcowden/cadquery #302: Plane offset issue after creating objects

bweissinger, 2018-11-28:
I have noticed a bug that seems to effect the Z-axis location of a plane after creating an object and then offsetting the plane, or creating a new plane on a face. The next object created will be in the correct location, but subsequent objects are displaced along the Z-axis of the plane.

It is a little hard to explain in words, so assume I run the following script.
```python
import cadquery as cq

result = cq.Workplane("XY").box(1, 1, 10).box(4, 4, 1).box(8, 8, 1)

show_object(result)
```
Also assume that I run the same script, but change result to be:
```python
result = cq.Workplane("XY").workplane(offset=5).box(1, 1, 10) \
    .box(4, 4, 1).box(8, 8, 1)
```
These two scripts produce the expected results.

[image]

Although it cannot be seen in the image, one is in fact offset 5 units in the positive Z direction.

Now assume I change the script so that the first box is created at the origin, then the workplane is offset by 5 units and the other two boxes are created. This should center both boxes at the end of the first one.
```python
result = cq.Workplane("XY").box(1, 1, 10).workplane(offset=5) \
    .box(4, 4, 1).box(8, 8, 1)
```

[image]

However, that is not the output. Instead, we get one box at the end of the first, and the other is offset in the negative Z direction from our workplane.

Also worth noting, if the Z length of the first box is changed to a much larger number,

```python
result = cq.Workplane("XY").box(1, 1, 1000).workplane(offset=5) \
    .box(4, 4, 1).box(8, 8, 1)
```

[image]

the center of the third box approaches [0, 0, 0] (in global coords). The top edge shows to be at 0.574 and the bottom at -0.426. It appears that it is behaving in a fashion similar to z = offset - (num/L)

This is also the case in plain CadQuery. If I run the following script in my python 2 virtual env with the CadQuery master branch
```python
import cadquery as cq

result = cq.Workplane("XY").box(1, 1, 1000).workplane(offset=5) \
    .box(4, 4, 1).box(8, 8, 1)

log = str(result.faces(">Z[2]").vertices().first().val().Z)

f = open('log.txt', 'w+')
print >> f, log
f.close()
```
I get `0.573891625616` as the Z coordinate of the selected vertices.

bweissinger, 2018-11-28:
After some testing, it seems that it is an issue with the center being moved. After the second box is created, the center of the plane is apparently moved to the center of mass of the CQ object. Calling `.center(0, 0)` after each box resets the center to the origin. 

I am not quite sure where this is hapening. [newObject just copies the plane](https://github.com/dcowden/cadquery/blob/ae8eaedf87cfb910b08f13f521aecbd71e260396/cadquery/cq.py#L1011) and then everything is returned, so I would guess that it is ocuring during the creation (union maybe) of the object.

jmwright, 2018-11-28:
> Not sure if this is a feature or a bug.

It's a feature. There's also a way to use the bounding box to have it act the way you expect. I'm not near a computer right now and won't be for awhile, but there should be more information in the tests. https://github.com/dcowden/cadquery/search?utf8=✓&q=boundingbox&type=

bweissinger, 2018-11-28:
Ok, danke. It makes sense, just not the behavior I initially expected. I thought it snapped to the center of the previously created object. That would be the case when there is only one prior object, such as
```python
result = cq.Workplane("XY").circle(1).extrude(1).circle(0.5).extrude(-1)
```
which places the bottom of the smaller cylinder face 0.5 units below the bottom of the larger cylinder. I didn't realize it took into account *everything*. I suppose the proper workflow in that case would be 
```python
result = cq.Workplane("XY").circle(1).extrude(1).faces("<Z").workplane().circle(0.5).extrude(-0.5)
```
I'll chock it up to user error and look the info on bounding box. Thanks!

closed 2018-11-28 by bweissinger

jmwright, 2018-12-03:
@bweissinger This can be counter-intuitive in some cases. In general I try to locate additional geometry based on previous faces rather than relying on centers. I feel safer doing that because if the other face moves, my new geometry will move along with it.