dcowden/cadquery #304: Getting "BRep_API: command not done" when adding fillet to object

rowanG077, 2019-03-17:
I need fillets for an object and keep getting this issues. I'm not sure how to work around it.

I have isolated the part where I get the issue. See the code below.

`# This is a CadQuery script template
# Add your script code below
import cadquery as cq
import math

# base specification
base_dia = 200
base_thickness = 10

# Feet specification
feet_length = 100
feet_side_length = 12.70
feet_angle = math.radians(80)
# cosine law
feet_diagonal1 = math.sqrt(2 * feet_side_length ** 2 - 2 * math.cos(feet_angle) * feet_side_length ** 2)
feet_diagonal2 = math.sqrt(2 * feet_side_length ** 2 - 2 * math.cos(math.pi - feet_angle) * feet_side_length ** 2)
feet_triangle_height = (math.sqrt(3) / 2) * feet_diagonal1
feet_fillet = 2

result = cq.Workplane("XY")\
    .polygon(3, base_dia)\
    .extrude(base_thickness)

# create feet
def create_feet_wire(self, l, angle):
    offset_y  = l * math.cos(angle / 2)
    offset_x  = l * math.sin(angle / 2)

    return self\
        .line(offset_x, offset_y)\
        .line(-offset_x, offset_y)\
        .line(-offset_x, -offset_y)\
        .close()

def create_feet_beam(self, length):
    tf = self\
        .transformed(
            offset = (0, base_dia/2 - feet_diagonal2/2 - feet_triangle_height, 0)
        )

    tf\
        .create_feet_wire(feet_side_length + 10, feet_angle)\
        .cutThruAll()

    return tf\
        .create_feet_wire(feet_side_length, feet_angle)\
        .extrude(-length, combine = False)

cq.Workplane.create_feet_wire = create_feet_wire
cq.Workplane.create_feet_beam = create_feet_beam

base_top_center_workplane = result.faces(">Z").workplane()

feet1 = base_top_center_workplane\
    .transformed(
        rotate = (0, 0, 30)
    )\
    .create_feet_beam(feet_length)

feet2 = base_top_center_workplane\
    .transformed(
        rotate = (0, 0, 150)
    )\
    .create_feet_beam(feet_length)

feet3 = base_top_center_workplane\
    .transformed(
        rotate = (0, 0, -90)
    )\
    .create_feet_beam(feet_length)

# union feet with base and add fillet
result = result.union(feet1).union(feet2).union(feet3)\
    .edges("|Z")\
    .fillet(feet_fillet)

show_object(result)
`

zignig, 2019-03-17:
If the fillet is larger than the smallest feature on your object the FreeCad fillet will fail.

rowanG077, 2019-03-17:
@zignig 

This is not the problem here. If I fillet every single edge(which includes the edges I want to fillet) on the part it works fine.

jmwright, 2019-03-17:
@rowanG077 Are you running CQ 1.0 (FreeCAD-based), or CQ 2.0 (PythonOCC-based)?

jmwright, 2019-03-17:
@rowanG077 your code runs fine for me in [CQ-editor](https://github.com/CadQuery/CQ-editor) using CQ 2.0.

adam-urbanczyk, 2019-03-18:
Works for me as well:

[image]

@rowanG077 do you have any issues with using CQ2.0? If further processing is required in FreeCAD you can always export your model in STEP format.

rowanG077, 2019-03-19:
Thank you for your quick responses, I am using FreeCAD. For some reason I cannot get CQ-editor to work with various import errors using anaconda. But I guess that is another issue.

This can be closed.

jmwright, 2019-03-19:
@rowanG077 Are you running CQ on a Mac by chance? We've had the most issues with CQ-editor on Mac OS. I'm hoping to find some time to test the latest changes later today.

What version of FreeCAD are you running?

rowanG077, 2019-03-19:
@jmwright I'm running on Ubuntu 18.04.1 LTS. I will try to reinstall anaconda completely and try again. Unfortunately I have now already removed FreeCAD because I removed anaconda. FreeCAD was recently installed (~1 month) with the newest version.

jmwright, 2019-03-19:
@rowanG077 Ubuntu 18.04 is my daily driver. CQ-editor and CQ 2.0 work well for me on 18.04. If you're interested in making the transition to CQ 2.0, we can help you through it. With CQ 2.0 being PythonOCC-based, it's opened up a lot of possibilities for new and improved features, so I feel it's worth making the change.

@adam-urbanczyk Does the CQ-editor readme have up-to-date installation steps?

jmwright, 2019-03-19:
Ok. I'm going to close this for now so that separate issues can be created as needed later.

closed 2019-03-19 by jmwright

adam-urbanczyk, 2019-03-19:
@jmwright  yes the installation instructions are updated. Please use conda version >= 4.6