dcowden/cadquery #307: how to iteratively loft wires in cadquery?

lalebarde, 2020-04-14:
I can loft a 2D shape from one plane to another like this:

```
import cadquery as cq
from jupyter_cadquery.cadquery import show
result = cq.Workplane("XY").rect(1,2).workplane()  \
     .transformed(offset=cq.Vector(0, -0.5, 1.0),rotate=cq.Vector(10, 0, 0)) \
     .rect(1,2).loft(combine=True)
show(result)
```
[image]

I would like to repeat the operation several times like this:

```
import cadquery as cq
from jupyter_cadquery.cadquery import show

wp = cq.Workplane("XY").rect(1,2).workplane()
result = None
for i in range(0,5):
    wp2 = wp.transformed(offset=cq.Vector(0, -0.5, 1.0),rotate=cq.Vector(10, 0, 0))
    if result == None:
        result = wp2.rect(1,2).loft(combine=True)
    else:
        nextpart = wp2.rect(1,2).loft(combine=True)
        result = result.union(nextpart)
    wp = wp2
show(result)
```

I have the following error:

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-54-37e88c00eca8> in <module>
      9         result = wp2.rect(1,2).loft(combine=True)
     10     else:
---> 11         nextpart = wp2.rect(1,2).loft(combine=True)
     12         result = result.union(nextpart)
     13     wp = wp2

~/anaconda3/envs/cq-jl/lib/python3.7/site-packages/cadquery/cq.py in loft(self, filled, ruled, combine)
   2897         self.ctx.pendingWires = []
   2898 
-> 2899         r = Solid.makeLoft(wiresToLoft, ruled)
   2900 
   2901         if combine:

~/anaconda3/envs/cq-jl/lib/python3.7/site-packages/cadquery/occ_impl/shapes.py in makeLoft(cls, listOfWire, ruled)
   1594         # the True flag requests building a solid instead of a shell.
   1595         if len(listOfWire) < 2:
-> 1596             raise ValueError("More than one wire is required")
   1597         loft_builder = BRepOffsetAPI_ThruSections(True, ruled)
   1598 

ValueError: More than one wire is required
```

At least, If I suppress the for loop and keep the same variable decomposition, it works:

```
import cadquery as cq
from jupyter_cadquery.cadquery import show

wp = cq.Workplane("XY").rect(1,2).workplane()
wp2 = wp.transformed(offset=cq.Vector(0, -0.5, 1.0),rotate=cq.Vector(10, 0, 0))
result = wp2.rect(1,2).loft(combine=True)
wp = wp2
show(result)
```

I think I am close to the result, but I cannot find where I mistake myself. The problem occurs when I come in the loop the second time. I think I have accumulated already two wires in wp2. I should keep only the last one before adding a new one, but I don't know how to do it.

jmwright, 2020-04-14:
@lalebarde Are you using CadQuery 1.0 or 2.0? For something like this it would be best to use 2.0 because that's where all the new development is being done now instead of this repo.

https://github.com/CadQuery/cadquery

I saw your question on the Google Group, but haven't been able to spend a lot of time with your sample code. Please post this to the new repo and hopefully we can find an answer for you.

lalebarde, 2020-04-14:
Thanks jmwright. Yes, I am using 2.0. ok, I post on the new repo, I did not catched it.

lalebarde, 2020-04-14:
re-opened on the new plateform [here](https://github.com/CadQuery/cadquery/issues/319)

closed 2020-04-14 by lalebarde