dcowden/cadquery #308: Installation process

alexmyself, 2020-05-12:
The installation process as describe on this page:

https://cadquery.readthedocs.io/en/latest/installation.html#installation

doesn't works, at least with fresh anaconda on a mac, but probably not only on mac, problems are dependencies versions oriented.

What works:
-install anaconda via their installer: https://www.anaconda.com/products/individual
-launch anaconda navigator
--create an environment with python 3.6 ( superiors versions fails at installing cadquery )
--click on the arrow of this new environment to "open terminal"
-paste in the opened terminal: conda install -c conda-forge -c cadquery cadquery=2

What doesn't works ( or have been met on my troubles path, consistency should be checked):
-install on win 10 32bit (... )
-anaconda environment > python 3.6
-installation on the  "base" (default) anaconda environment (python 3.7 atm)
-installation of PythonOCC prior to cadquery (something like "ca-certificates" version became a problem).

Which error messages have been seen and totally disappeared once environment 3.6 has been used:
-Solving environment: failed with initial frozen solve. Retrying with flexible solve.
-Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source. 
-Found conflicts! Looking for incompatible packages.
-UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:
-The following specifications were found to be incompatible with each other:
-Package libcxx conflicts for:

Hopes this helps, thanks for cadquery! :)

dcowden, 2020-05-12:
@alexmyself  HI, thanks for reporting this!
Have you checked out cadquery 2.0 over at https://github.com/CadQuery/cadquery? 
It's newer, and i think has more stable conda packages.

alexmyself, 2020-05-12:
The version 2.0 is what i've followed, precisely:

-github of v2.0 as you suggested: https://github.com/CadQuery/cadquery
-then the first link to the doc, at the top of the github page: https://cadquery.readthedocs.io/en/latest/ 
-then the link to installing cadquery: https://cadquery.readthedocs.io/en/latest/installation.html

I've just installed the cq-editor, it works fine (...it starts and looks like "normal" at the instant, can't says it really "works fine" but installation went well and it seems to act like it should )

From my user point of view you should just state on the installing cadquery page:
-install anaconda
-launch anaconda navigator
-make an environment with python 3.6
-open the terminal for the two commands:
-- conda install -c conda-forge -c cadquery cadquery=2
-- conda install -c cadquery -c conda-forge cq-editor
-start editor by this terminal: cq-editor

From my user point of view what is (..a bit) misleading:
-@https://cadquery.readthedocs.io/en/latest/installation.html:
--Anaconda or Miniconda (Python 3.x editions), Python 3.x
--> say python 3.6  ( as anaconda propose 3.8 for creating new environment, which looks likes "the good choice" )
--CadQuery requires PythonOCC and Python version 3.x
--> says ..nothing, just not mention PythonOCC on installing page.. it is installed in a right way with your command and maybe provide problems if installed separately before cadquery.

Have I said thanks for cadquery? :)

jmwright, 2020-05-13:
@alexmyself please post this on the CadQuery 2.x issue tracker since this is related to CQ 2.x and not CQ 1.x. We've migrated away from this repo.

https://github.com/CadQuery/cadquery/issues

closed 2020-05-17 by jmwright