dcowden/cadquery #309: Can't sweep a path along a helix

nopria, 2020-06-23:
Hi, I'm trying to make a helicoidal spring with CadQuery version 2.0, but when I try to sweep the circle along an helix I got the error `'Wire' object has no attribute 'wire'`.

```
import cadquery as cq
thread_profile = cq.Workplane("XZ").circle(5)
thread_helix = cq.Wire.makeHelix(50,50,100)
show_object(thread_helix)
r = thread_profile.sweep(thread_helix)
show_object(r)
```

Even if there is a good programmatic reason for this, I think that from the user perspective the error is quite confusing, shouldn't a `makeHelix` method return a `wire`?
In an old FreeCAD discussion I found a way to convert a `Wire` helix to a `wire`, using it I get no more error but the result is quite strange and not correct ([the circle is swept along a path different from the helix](https://github.com/jmwright/cadquery-freecad-module/issues/122#issuecomment-647629317)).

adam-urbanczyk, 2020-06-23:
Please post in the CQ issue tracker: https://github.com/CadQuery/cadquery .

There is no type `wire` in the CQ type system. You need to wrap your `Wire` object into a `Workplane`. The behavior you are calling strange is correct. You need to put the circle in the desired place. Depending on its location the sweep result will look differently:
```python
import cadquery as cq

R = 50
r = 5
N = 5

thread_profile = cq.Workplane("XZ",origin=(R,0,0)).circle(r)
path = cq.Workplane("XY", obj=cq.Wire.makeHelix(R,N*R,R))

res = thread_profile.sweep(path)
```
[image]

nopria, 2020-06-23:
Thank you for answering to my wrongly addressed issue, which in addition was unfounfed since the initial position of circle was wrong, However, the fixed original code works as espected, whereas with your cleaner code I get the error `'Vector' object has no attribute 'ShapeType'` in cq-editor.

adam-urbanczyk, 2020-06-23:
Yup, I wasn't using master branch.

closed 2020-06-23 by nopria