dcowden/cadquery #310: Consider marking this repo as archived

adam-urbanczyk, 2020-06-23:
Maybe it'd be good to mark this repo as archived (see https://github.blog/2017-11-08-archiving-repositories/ ) to prevent confusion and issues being open in the wrong place

jmwright, 2020-06-23:
@easyw do you have thoughts about archiving this? I know you have built a KiCAD toolchain that includes CQ 1.x.

adam-urbanczyk, 2020-07-21:
Any updates/thoughts?

easyw, 2020-07-22:
@jmwright 
AFAIK a lot of kicad &freecad users are using your (@jmwright) nice CQ WB...
Would this archiving be related/affect your WB?

jmwright, 2020-07-22:
@easyw I think it would block me from eventually fixing [this issue](https://github.com/jmwright/cadquery-freecad-module/issues/147) unless I fork the repo. That code has been fixed in CQ 2.x for Python 3.8, but hasn't been fixed in CQ 1.x.

hkienle, 2021-01-04:
Adding to @easyw that "a lot of kicad &freecad users are using your (@jmwright) nice CQ WB" there is a scientific paper from December 2019 that mention the workbench: https://journals.plos.org/plosone/article?id=10.1371/journal.pone.0225795

adam-urbanczyk, 2021-01-04:
So what is the proposal here? Is there anyone willing to maintain CQ1?

dcowden, 2021-01-04:
I'm definitely in favor of archiving dcowden/cadquery. If there is a need for small fixes to be applied to the old version, perhaps those could be done on a fork of dcowden/cadquery?

jmwright, 2021-02-08:
I have cleaned up a few things this repo, including pushing an updated readme to PyPi warning people that they need to move to CQ 2.x. I also added a deprecation warning when the cadquery 1.x module is imported. I think that should allow us to go ahead and archive this.

@dcowden I think you're the only one who has access to the Settings in order to archive this. Since you started this whole project, it seems fitting that you would be the one to archive this repo anyway.